tobyweston/temperature-machine #2: How to change sensor name?

BoomerBrian, 2017-01-25:
How do we change the sensor names? For example how to change garage-sensor-1 to garage-fermentation.

tobyweston, 2017-01-26:
This isn't configurable at the moment. It's based on the 'host' name, so you'll get `garage-sensor-1` and `garage-sensor-2` for now.

However, as part of issue #1 I'm thinking about implementing some sort of configuraiton to map individual sensor names (like `28-xxxx`) to whatever you like.

How would you like to see that mapping? A file? an environment variable? something else? Not supplying one would likely default to existing behaviour.

BoomerBrian, 2017-01-26:
Hello Toby.

Thanks for the response.  I would prefer a configuration file.

tobyweston, 2017-01-26:
If you have just one sensor, you can use your hostname. Not sure if that was clear, but if you change your host name to "fermentation" it'll be the default for all the sensor names and show up on the UI.

BoomerBrian, 2017-01-26:
Good idea. Thanks.

tobyweston, 2017-02-03:
#1 has had some love so (on a branch) the sensor names are now available. 

All we'd need to do is add a mapping layer from sensor name to friendly name and plumb it all in.

I'll chip away but welcome PRs.

kiwamacey, 2017-07-05:
I  would also find being able to change the sensor names useful. Have installed 5 sensors on one Pi3 in different rooms to monitor how well the HRV is working.  This project seems to be the easiest and best option I have found, Great work!

tobyweston, 2017-07-05:
I'll bump this to the next thing I look at... suggestions as to how welcome

kiwamacey, 2017-07-06:
some kind of config file file would be my suggestion, something like 
28-000009824226 = Bedroom1
28-000009824227 = Kitchen
28-000009824228 = Outside
space, comma, tab separation what ever is easiest I guess.
My other question would be how can i change the colors on the graph? I am a bit colour blind and the defaults I have for sensors 1,4 and 5 (particularly 1 and 4) are much the same to me. if you can point me to the settings i don't mind rebuilding it.

tobyweston, 2017-07-06:
@kiwamacey you can tweak [`Graph`](https://github.com/tobyweston/temperature-machine/blob/83175f2e572c61ce3a047711201fa6612b0e6aa6/src/main/scala/bad/robot/temperature/rrd/Graph.scala) for the PNGs at the bottom on the UI and the colours for the JavaScript main chart are actually in the companion / UI project [https://github.com/tobyweston/temperature-machine-ui](https://github.com/tobyweston/temperature-machine-ui).

It's a little more involved changes these are that project has it's own build process which copies minified JavaScript into the `resources` folder of this project. Where as this project uses `sbt` for building, the UI project uses `npm`. If those things mean nothing to you, I'd avoid it. If they do, happy hacking.

Failing that, you could try looking for the colours in the pre-built `resources` folder, in `main-xxx.js` but it's minified and horrible to look at.

I'll have a think about adding configurable colours, or if you wanted to send over some more contrasting colours, I'd be happy to replace what's there.

kiwamacey, 2017-07-26:
Thanks for that, I have replaced the purplish colour for sensor 4 with an orangish colour (#db843d) and it works well for me.I did as you suggested and found the colours, replaced the fourth one and rebuilt with the original sbt command.

tobyweston, 2017-07-26:
Great, if you want to do a pull request, I'll roll it in. Otherwise, I'll make the same change when I get to it. Cheers

kiwamacey, 2018-08-15:
Hi Again, I hope you can help again. I have just upgraded the source on my b+ server. The new changes in the doccy's don't line up with the build from source docs. I compile the souce as th pi user which does not install the temperature-machine file to the executable path. I could not find it with the find command under the code directory. the start.sh script does not start program correctly either. I am using the source up as I have edited the old Graph.scala and main***.js for my colour blindness with colours I can make out better. I was going to finally upgrade and see if you wanted to include them in working release.does the new version have the start-server and start-client .sh files, the git pull version i did does nt?

tobyweston, 2018-08-15:
Hey @kiwamacey sorry about that! The docs should be consistent, make sure you're looking at either

1. http://temperature-machine.com/docs/getting_started/installing.html
1. http://baddotrobot.com/blog/2016/03/23/homebrew-temperature-logger/

We recently change the recommended installation method from building from source to install via `apt`. It's quicker for you and now runs a service that restarts automatically and other good stuff. However, any changes you've made will be lost.

We can do a couple of things:

1. Get your changes into the main project so you can install and update via `apt` (my preference)
1. Get you going building and deploying from source

For the first one: did you fork my repo and make your changes? That's the easiest way for me to see what you changed and potentially work it into the app.

The second one should be doable and the instructions online should be accurate - we prob just have to figure out what's wrong.

kiwamacey, 2018-08-18:
Thanks for the reply,
I have just forked and replaced the two files Mentioned previously. main.4b***.js and Graph.scala. I hope the colours aren;t to gaudy for every one, they work for me but may not work for someone with the different or a worse case of colour blindness. I have used Black at the beginning of the string in main*.js and at the end of Graph.scala as I have just added a last detector on a annother pi for my garage ( this seems to give a better reading than ones install on the sofette of my house). The garagepi/detector is first on the main html page but last on the daily, weekly and mothly graphs.
still had issues build from source. I thingk the only thing I havent changed is the SD card. my last attempt was under home/pi/fork/temperature-machine, failed with the error mentioned at the bottom of the instructions 
(/usr/local/bin/sbt: line 3:  6014 Killed                  java $SBT_OPTS -jar `dirname $0`/sbt-launch.jar "$@" or different PID)
Did this build while the old version was still live and noticed wome weird things - temperatures on the webpage later, temps went up to 110 Deg C at about the time I ran the build??!!
Chat Soon, Thanks

tobyweston, 2018-10-03:
It is yes, just haven't much time to spend on the project atm 😢

DaddyDeej, 2022-06-15:
@tobyweston.. Wondering if this was ever added? Thanks!

Aupirat, 2023-01-11:
I would be very interessed in that feature as well.
I am running 5 sensors on a PI-Zero with my zentral heating unit. It's running great. But the above discussed name mapping feature would be very helpfull so that my other family members can read the data as well. Is there a known way to manipulate the names in the meantime? Thank you! 
[image]

DemBones78, 2023-01-11:
> Hey,
> This is a dead project. You will get no help and no updates.
>
> Please take me advice and look in to home assistant.   Hass.io
>
> With a espd1 mini a 2$ part your probes will be ruining great on home
> assistant your can have stunning graphs via grafona. And do all sorts of
> automation with that sensor data
>
>
> Watch a few videos about home assistant
>
> Then watch a video about esphome
> The party that let's you build cheap esp devices

here is some more information

[image: wemos-mini-and-ds18b20_bb-1.webp]

http://www.esp8266learning.com/wemos-mini-ds18b20-temperature-sensor-example.php
the one wire temp probs hook up the same way

this is what grafana looks like, this is from my greenhouse. it has many
probs

[image: Capture.JPG]

this is how it can look in a very basic dashboard
[image: Capture.JPG]

this is what the webpage of the esp look like , its setup automatically via
the esphome config all you do is say webpage yes this is the default

[image: Capture.JPG]

this stuff is really easy to set up there is lots of support, the
Homeassistant board is really good, plus there is discord and lots of
videos and web tutorials

this is a great thing to learn about because with this project you can do
damn near anything.  not just a few temp probs and no automation.

that and you can get esp chips in all sorts of devices and replace the
software to esphome or tasmota so it runs on your cloud and some internet
company

DemBones78, 2023-01-11:
oh and by the way changing names in home assistant is very easy

on the devices, they look like this

# dallas sensors
dallas:
   pin: GPIO5
   update_interval: 10s
# Individual sensors
sensor:
  - platform: dallas
    address: 0x17000004FD7F2A28
    name: "6 Temp-OnBoard"
  - platform: dallas
    address: 0x12000004FBD82828
    name: "2 Temp-High"
  - platform: dallas
    address: 0xD33C01D6074F0428
    name: "0 Temp-low"
  - platform: dallas
    address: 0x42000005DCB47628
    name: "4 Temp-Outside"
  - platform: dallas
    address: 0x6B3C01D607A0EB28
    name: "5 Temp-Rad"
  - platform: dallas
    address: 0x963C01D607FE4228
    name: "1 Temp-Water"
  - platform: dallas
    address: 0x57000005B4B0E728
    name: "3 Temp-Mid"
  - platform: dallas
    address: 0xcb3c01f096287128
    name: "7 Chimmey"

just edit the name, in-home assistant you can use aliases for devices that
may change function over time.
For example, i have some smart switches that get moved around from
controlling Christmas lights to fans to pumps to lamps ect. i just change
the aliases

HenrysCat, 2023-01-11:
> oh and by the way changing names in home assistant is very easy on the devices, they look like this # dallas sensors dallas: pin: GPIO5 update_interval: 10s # Individual sensors sensor: - platform: [...]

And that all looks soooooo easy to set up, I bet anyone could do it in a few minutes with no previous knowledge at all.

DemBones78, 2023-01-11:
Exactly!

DemBones78, 2023-01-11:
https://esphome.io/components/sensor/dallas.html

oh look tons of support on an active project.  its all temples you really
just edit the small parts , its easy and I cant code any thing

Aupirat, 2023-01-12:
Ok, thank you for the information.
But I was specifically looking for a temperature logger solution that uses the PI-Zero and the DSB1820 locally and this here works great. I was hopping that somebody had found a simple workaround or tweak to solve the naming issue.

Thank you for the hint to "home assistant" that looks like a thing I will try. Unfortunately they kicked out the direct GPIO support in 06-2022 and now you have to look for addons that allow you to get the data from 1-wire sensors.
Still, i am going to try that on a separate SD card with my Pi-Zero....

Thanks agin for the quick response!

DemBones78, 2023-01-12:
you don't need an add on you just use ESP devices for the sensors. like this
https://www.amazon.com/Organizer-ESP8266-Internet-Development-Compatible/dp/B081PX9YFV/ref=sr_1_5?keywords=esp8266&qid=1673531889&sr=8-5

that reports to your homeassiatant

HenrysCat, 2023-01-13:
Lots of vague useless inforfmation there, how does one connect to the ESP device to add the code? how do you connect it wifi network? are they assigned an IP like a R-Pi?

HenrysCat, 2023-01-24:
Finally did it, thanks for pointing me in the right direction DemBones78, I used this guide https://microcontrollerslab.com/esp32-esp8266-dht11-dht22-influxdb/ and now have my readings in Grafana.
Does seem daunting at first, and it did take me 3 days to figure it all out lol

DemBones78, 2023-03-16:
Glad to hear it!