tobyweston/temperature-machine #13: Negative temperatures

blinkingbee, 2017-06-09:
Problem with temperatures below zero being logged as large positive numbers (e.g. 4085.3 °C). This is also reflected in the graphs auto-scaling to 4000 °C. 

When I read the sensor temperature using a python script with an expression such as  `echo `/usr/local/bin/get_temp.py | head -1 | awk -F. '{print $1"."int(($2/100)+0.5)}'`` this returns the correct negative temperature. 

Could this be a problem with how Temperature-Machine reads the sensor and handles signing?

[image]

[get_temp.py.zip](https://github.com/tobyweston/temperature-machine/files/1064017/get_temp.py.zip)

tobyweston, 2017-06-09:
Any chance you could post a couple of example files (e.g. from ```/sys/bus/w1/devices/28-0416a443d6ff/w1_slave```)?

blinkingbee, 2017-06-09:
Samples taken at 3 different temperatures with icepack from my freezer:
[w1_slave samples.zip](https://github.com/tobyweston/temperature-machine/files/1065257/w1_slave.samples.zip)

It would be very useful if you could use the pre-calculated value from the scratchpad and thus exchange some precision for valid temperatures below zero.

blinkingbee, 2017-06-09:
Posted some examples for both +ve and -ve temperatures

blinkingbee, 2017-06-12:
I have updated Parser.scala and ParserTest.scala in both client and server but temperatures below zero still display as large positive numbers. Is there anything else I need to do?

tobyweston, 2017-06-12:
I haven't deployed myself yet so am still testing. I'll prepare a 'package' once done and close the issue here.

How are you deploying (see http://temperature-machine.com/docs/getting_started.html#setup-the-software).

If you're building from source, you can test by following [these instructions](http://temperature-machine.com/docs/getting_started/build_from_source.html) having first done the following;

```
$ git pull
$ git checkout issue_13
```

blinkingbee, 2017-06-12:
Thanks, maybe I jumped the gun here. I originally deployed using the pre-installed image. If I build from source does this just overwrite the original temperature-machine?

tobyweston, 2017-06-12:
Yep, on the pre-installed image under `~/code/temperature-machine`. Just run the above and rebuild the `.jar` file (steps 2. from [these instructions](http://temperature-machine.com/docs/getting_started/build_from_source.html).

So basically,

```
cd ~/code/temperature-machine
git pull
git checkout issue_13
sbt -J-Xmx512m -J-Xms512m assembly
```
Then restart.

__Note__ that this will checkout the _branch_ so once it's merged back into `master`, you'll have to reverse the procedure (i.e, `git checkout master` then `sbt assembly`). I'll merge back after testing and building a new image etc (which will prob take a little while)

Let me know how you get on.

closed 2017-06-12 by tobyweston

tobyweston, 2017-06-12:
Just merged bf10a84 to master so ignore the `git checkout issue_13` step...

blinkingbee, 2017-06-13:
I have rebuilt both master and slave as described above but am still getting large positive numbers when temperature goes below zero. Separately monitoring the sensor scratch file pre-calculated value shows the correct negative temperature.

tobyweston, 2017-06-13:
and the `sbt assembly` step worked for you? Sometimes it crashes on the Pi because of the limited resources and you have to re-run. Check the timestamp of the `temperature-machine.jar` file. Something like `find . -name "*.jar"` from the temperature-machine fold should locate it. Make sure it's been updated.

blinkingbee, 2017-06-13:
`-rw-r--r-- 1 pi pi 28515459 Apr  2 14:09 temperature-machine-2.0.jar` It looks like it's not been updated so I'll try a rerun

tobyweston, 2017-06-13:
Cool, when you run `sbt assembly` (on the Pi zero in particular), it can fail more or less silently. Re-running usually works (but __don't__ `sbt clean` first).

blinkingbee, 2017-06-13:
Mostly success. My two client units built successfully and the one I've tested displays negative temperatures correctly. However the server returns errors and won't build. I have attached the console output for your comment. I may have to start over with the server if I can't solve it.

[attempted build.txt.zip](https://github.com/tobyweston/temperature-machine/files/1071035/attempted.build.txt.zip)

blinkingbee, 2017-06-13:
I have fixed server by re-installing from image then rebuilding from git. Great to see 24h time in main graph now and I see you have removed reference to sensor serial numbers too. Thanks for your help.

tobyweston, 2017-06-13:
Great stuff, glad its working for you! (BTW, if you have more than one sensor attached, it'll display the serial numbers again). 😄

commit bf10a848 (2017-06-12): fixes #13 support negative temperatures