tobyweston/temperature-machine #16: pi zero with eth0 nic

lhalep, 2017-10-07:
Hi,
I have pi zero with eth0 usb nic and following the guide when run the  ./start.sh i got : 
wlan0: error fetching interface information: Device not found
Started your temperature-machine, redirecting output to temperature-machine.log, PID stored in temperature-machine.pid

i also got the following on the log file : 
root@pizero:~/code/temperature-machine# more temperature-machine.log
Error: Unable to access jarfile target/scala-2.12/temperature-machine-2.0.jar

also at the end of the sbt assembly I got:
   at java.lang.Thread.run(Thread.java:745)
[error] (temperature-machine/compile:compileIncremental) java.lang.ClassNotFoundException: xsbt.CompilerInterface
[error] Total time: 315 s, completed 07-Oct-2017 14:03:27

on the instructions there was no mention to java installation. This is a fresh install of raspbian lite on a pi zero. so in order to install the java I run : 
root@pizero: sudo apt-get install oracle-java8-jdk
and the result of the java ver is the following : 
root@pizero:~/code/temperature-machine# java -version
java version "1.8.0_65"
Java(TM) SE Runtime Environment (build 1.8.0_65-b17)
Java HotSpot(TM) Client VM (build 25.65-b01, mixed mode)

too much? (should have split into 3 issues?)
Looking forward for your input

Laz

tobyweston, 2017-10-07:
Have you testing the wireless out on the machine? the `wlo0: error fetching interface information` sounds like a general wireless error unrelated to temperature machine. Does [this help](https://unix.stackexchange.com/questions/345356/ifconfig-wlan0-shows-fetching-interface-information-device-not-found/369080)?

The unable to access jarfile issue: does the `temperature-machine-2.0.jar` file exist in `/home/pi/code/temperature-machine/target/scala-2.12`? If it doesn't you need to get it there (it could be that the SBT process failed if you're compiling manually and so it wasn't created). If it does, check the permissions, it should look like this:

```
-rw-r--r-- 1 pi pi  28M Jun 22 07:28 temperature-machine-2.0.jar
```

Make sure you login as `pi` user and try again manually with:

```
cd target/scala-2.12
java -Xmx512m -jar temperature-machine-2.0.jar 
```

RE the SBT error; lets try the above first and go from there. Can you explain the steps you took, step by step so I know which method of installation you're trying. Did you `git pull` or are you working from a download?

lhalep, 2017-10-08:
amazing your immediate response, thank you
regarding the wireless, the pi zero that i have does not have an onboard wireless (it is the first version of pi zero)

that is why i have a usb ethernet nic

lhalep, 2017-10-08:
as an effect to previous , my pi zero has only eth0 interface

lhalep, 2017-10-08:
regarding to other issues I will investigate and come back to you

tobyweston, 2017-10-08:
Spawning a new issue #17 for the wireless issue

lhalep, 2017-10-08:
regarding the installation process,
on a fresh raspbian pi zero, 
I followed the guidelines from this:

http://baddotrobot.com/blog/2016/03/23/homebrew-temperature-logger/

but I just noticed that before I start, I elevated my privileges (su root)
so everything I did was as user root and not as user pi (there was no mention on the guidelines to be user pi when installing)

so when on the guide it was said : 
mkdir ~/code

it was on the path : 
/root/code

also I search the entire disk with
find / -type f -iname "temperature-machine-2.0.jar"

and came up with nothing

tobyweston, 2017-10-08:
There's some more comprehensive instructions on the site at http://temperature-machine.com/ (click Documentation in the top right). 

The recommendation is not to do things as `root`, stick to `pi`.

tobyweston, 2017-10-08:
There's some more comprehensive instructions on the site at http://temperature-machine.com/ (click Documentation in the top right). 

The recommendation is not to do things as `root`, stick to `pi`.

When compiling on the Pi, it will very likely crash because of memory / general Pi weirdness. Re-running usually sorts it out but make sure the `jar` file has been generated. If you can't see the `jar` file after running `sbt -J-Xmx512m -J-Xms512m assembly`, then it most likely crashed.

lhalep, 2017-10-09:
so what now? should uninstall and reinstall as pi?
if uninstall i should just delete folders?

lhalep, 2017-10-09:
so if i have only one pi, i should run server and client? (following the Download Binary Release instructons)

tobyweston, 2017-10-09:
@lhalep raised #18, is this what you see?

tobyweston, 2017-10-10:
> so if i have only one pi, i should run server and client? (following the Download Binary Release instructons)

No, just run the server.

tobyweston, 2017-10-30:
closing issue as I think #17 covers the main issue

closed 2017-10-30 by tobyweston