tobyweston/temperature-machine #19: can't seem to make it work, need help

lhalep, 2017-10-10:
Hi again,
ok so the eth0 is ok
so i started again. removed all directories and started again as user pi

followed the most detailed instractions from
http://baddotrobot.com/blog/2016/03/23/homebrew-temperature-logger/

1)I followed the steps to setup sbt
2)Created the file /usr/local/bin/sbt
3)cloned the data logger’s from Git repository
4)changed the .sh files to reflect my eth0 instead of wlan0
5)build the binary

during the build I noticed

SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.

at the end of the build of binary I got
[info] Done updating.
[info] Compiling 48 Scala sources to /home/pi/code/temperature-machine/target/scala-2.12/classes...
[info] 'compiler-interface' not yet compiled for Scala 2.12.3. Compiling...
[info]   Compilation completed in 220.088 s
[info] Compiling 26 Scala sources to /home/pi/code/temperature-machine/target/scala-2.12/test-classes...
/usr/local/bin/sbt: line 3:  2222 Killed                  java $SBT_OPTS -jar `dirname $0`/sbt-launch.jar "$@"
pi@pizero:~/code/temperature-machine $ 

i didn't know if all was ok or not, but then I run

lhalep, 2017-10-10:
going back to your replies ... I am trying the 
sbt -J-Xmx512m -J-Xms512m assembly
approach since the target/scala-2.12/temperature-machine-2.1.jar is nowhere to be found
(I haven't deleted any previous installation folders...hmm)

I will let you know...

tobyweston, 2017-10-10:
It'd be helpful if you stick to the thread/issue that we've created. I'll close this one for now.

The like that says `Killed java $SBT_OPTS -jar dirname $0/sbt-launch.jar "$@"` is saying that the sbt process was killed. It crashed (which is common when running on the Pi) and the fact that there is not jar file confirms it.

You would need to re-run until the jar is created. Confirm the jar exists before attempting to run `start.sh`. That is failing because (as it says), it can not access the jar file (because its not there).

closed 2017-10-10 by tobyweston