tobyweston/temperature-machine #20: Error message: This site can't be reached

Cloudedhouse, 2017-10-15:
I've tried this on a Pi B, that should work shouldn't it?

It was all going so well until the very last step ... IPaddress:11900 This site can't be reached.

port 11900 doesn't appear to be open.

Any ideas anyone?

Many thanks

singh763173, 2017-10-28:
+1 on this.

Followed all the steps to a T but not presented with the landing page when pointing to port 11900 in the browser.

Pi Zero W User

tobyweston, 2017-10-28:
Can you post the logs?

`cat ~/code/temperature-machine/temperature-machine.log`

You followed the instructions from http://temperature-machine.com/docs/ ? Which method were you using?

How are you starting it up? Are you adding the start command to `/etc/rc.local` or running it directly from the shell? What output do you get when you run it from the shell? The start command should be something like `/start-server.sh garage bedroom` where `garage` and `bedroom` are the target RRDs.

Cloudedhouse, 2017-10-28:
Toby

Thanks for you help on this, it's a while since I set this up so I can't quite remember the details

I followed these instructions http://baddotrobot.com/blog/2016/03/23/homebrew-temperature-logger/

This is the message I get 
pi@raspberrypi:~ $ cat ~/code/temperature-machine/temperature-machine.log
Error: Unable to access jarfile target/scala-2.12/temperature-machine-2.1.jar

Thanks again

CH

tobyweston, 2017-10-28:
Ok, well there's some more up-to-date / canonical docs on the site I mentioned. If you'd had a go from the blog instructions, you'd tried to build from source. There's other options now (i.e. get a pre-installed distro). Re-read the instructions on `Getting Starting > Building from source` on the site in case anything changed.

I suspect the compile stage failed for you and you don't have the `jar` file it needs. Check with `ls -la ~/code/temperature-machine/target/scala-2.12` and see if there's any JAR files there.

Anyway, stick with what you've got and do the following.

```
cd ~/code/temperature-machine
git pull
time sbt assembly 
```

You might try adding `-J-Xmx512m` if you have memory problems.

SBT will fail **a lot** on the Pi, I think due to lack of resources so you may need to run it several times (2 usually works for me) to actually get it to generate the JAR file. Look for errors like #19 and terminated processes and don't continue until you have the JAR generated in the `target` folder. No JAR = No temperature-machine.

singh763173, 2017-10-28:
Hi Toby - I built from source from the blog and the new site you linked.

I found that I had to run sbt assembly twice but i can confirm i have the temperature-machine-2.1.jar file in the scala2.12 folder. @Cloudedhouse you may want to try this if you haven't already

I've added the output from the logs below:

`pi@TempMonitor01:~/code/temperature-machine $ cat ~/code/temperature-machine/temperature-machine.log
Starting temperature-machine (server mode)...
RRD initialising for 'TempMonitor', 'TempMonitor' (with up to 5 sensors each)...Exception in thread "main" java.lang.IllegalArgumentException: Datasource already defined: DS:TempMonitor-sensor-1:GAUGE:35:U:U
	at org.rrd4j.core.RrdDef.addDatasource(RrdDef.java:219)
	at bad.robot.temperature.rrd.RrdFile.$anonfun$create$2(RrdFile.scala:38)
	at scala.collection.immutable.Range.foreach$mVc$sp(Range.scala:156)
	at bad.robot.temperature.rrd.RrdFile.$anonfun$create$1(RrdFile.scala:37)
	at bad.robot.temperature.rrd.RrdFile.$anonfun$create$1$adapted(RrdFile.scala:37)
	at scala.collection.immutable.List.foreach(List.scala:389)
	at bad.robot.temperature.rrd.RrdFile.create(RrdFile.scala:37)
	at bad.robot.temperature.task.Tasks$.$anonfun$init$3(Tasks.scala:20)
	at bad.robot.temperature.task.Tasks$.$anonfun$init$3$adapted(Tasks.scala:19)
	at scalaz.concurrent.Task.$anonfun$map$3(Task.scala:42)
	at scalaz.concurrent.Task$.Try(Task.scala:457)
	at scalaz.concurrent.Task.$anonfun$map$2(Task.scala:42)
	at scalaz.$bslash$div.flatMap(Either.scala:135)
	at scalaz.concurrent.Task.$anonfun$map$1(Task.scala:42)
	at scala.Function1.$anonfun$andThen$1(Function1.scala:52)
	at scala.Function1.$anonfun$andThen$1(Function1.scala:52)
	at scalaz.concurrent.Future.$anonfun$flatMap$1(Future.scala:58)
	at scalaz.concurrent.Future.step(Future.scala:118)
	at scalaz.concurrent.Future.unsafePerformListen(Future.scala:75)
	at scalaz.concurrent.Future.unsafePerformAsync(Future.scala:159)
	at scalaz.concurrent.Future.unsafePerformSync(Future.scala:184)
	at scalaz.concurrent.Task.unsafePerformSync(Task.scala:99)
	at bad.robot.temperature.ds18b20.SensorFile$.findSensorsAndExecute(SensorFile.scala:28)
	at bad.robot.temperature.server.Server$.delayedEndpoint$bad$robot$temperature$server$Server$1(Server.scala:61)
	at bad.robot.temperature.server.Server$delayedInit$body.apply(Server.scala:13)
	at scala.Function0.apply$mcV$sp(Function0.scala:34)
	at scala.Function0.apply$mcV$sp$(Function0.scala:34)
	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
	at scala.App.$anonfun$main$1$adapted(App.scala:76)
	at scala.collection.immutable.List.foreach(List.scala:389)
	at scala.App.main(App.scala:76)
	at scala.App.main$(App.scala:74)
	at bad.robot.temperature.server.Server$.main(Server.scala:13)
	at bad.robot.temperature.server.Server.main(Server.scala)`

I ran the curl command and got nothing.

Thanks in advance.

tobyweston, 2017-10-28:
@singh763173 it looks like you're trying to create two RRDs with the same name, `TempMonitor`.

Is that the name of your machine? Are you running just one machine? What's your start-up command? Have you got `TempMonitor` anywhere twice?

FYI `curl` will return nothing as the server isn't starting up. Good start verifying the JAR file actually exsits but you'll need to look for the following in the log file to verify the server has actually started. Once you see this, you should be good to go.

```
HTTP Server started on http://127.0.1.1:11900
```

singh763173, 2017-10-28:
@tobyweston bingo!

I had  su pi -c 'cd /home/pi/code/temperature-machine && ./start-server.sh TempMonitor01 TempMonitor02 &' in my rc.local file (i intended on having two devices but only one at the moment)

I've just removed the second hostname and it's now working!

Thank you for your help with getting this working - greatly appreciated!!

Probably for another ticket but is it straight forward to add a graph with a years view?

tobyweston, 2017-10-28:
Great stuff! 🥇 👍 

There is a character limit imposed by RRD which I think is 11 characters, that would mean your names got truncated to duplicates. If you add a machine later, you'll need a more unique name.

It's a good idea to put the RRDs in place now though. If you add `study`, `garage`, `outside`, `bedroom` whatever now but only use one, it'll mean you wont have to delete the RRD (basically the `~/.temperature` folder in the future. If you don't and only setup one RRD now and then want to add another later, you'll have to blast the folder. It's a constraint on the RRD library I'm using I'm afraid :(

tobyweston, 2017-10-30:
@singh763173 sounds like you're good to go, how about you @Cloudedhouse? Any joy?

Cloudedhouse, 2017-10-31:
No sorry Toby, SBT has failed for me probably 10 - 12 times. Pasted below is the last of 100's of error messages.

Perhaps a Pi B isn't man enough?

.........................................................................................................................................
[warn] Strategy 'discard' was applied to a file
[info] Assembly up to date: /home/pi/.sbt/0.13/staging/695479af9da4737401a3/temperature-machine/site/target/scala-2.10/site-assembly-0.1-SNAPSHOT.jar
[error] Error: Total 127, Failed 0, Errors 25, Passed 102
[error] Error during tests:
[error]         bad.robot.temperature.server.HttpServerTest
[error]         bad.robot.temperature.server.TemperatureEndpointTest
[error]         bad.robot.temperature.rrd.RrdDbOpsTest
[error]         bad.robot.temperature.server.EitherToResponseTest
[error]         bad.robot.temperature.ErrorOnTemperatureSpikeTest
[error]         bad.robot.temperature.rrd.XmlTest
[error]         bad.robot.temperature.rrd.HostTest
[error]         bad.robot.temperature.PercentageDifferenceTest
[error]         bad.robot.temperature.client.HttpUploadTest
[error]         bad.robot.temperature.ds18b20.SensorReaderTest
[error]         bad.robot.temperature.IpAddressTest
[error]         bad.robot.temperature.ds18b20.ParserTest
[error]         bad.robot.temperature.MeasurementTest
[error]         bad.robot.temperature.RecordTemperatureTest
[error]         bad.robot.temperature.IgnoreTemperatureSpikesTest
[error]         bad.robot.temperature.server.DiscoveryServerTest
[error]         bad.robot.temperature.task.SchedulerTest
[error]         bad.robot.temperature.rrd.ChartJsonTest
[error]         bad.robot.temperature.server.ConnectionsEndpointTest
[error]         bad.robot.temperature.SensorReadingTest
[error]         bad.robot.temperature.ds18b20.SensorFileTest
[error]         bad.robot.temperature.TemperatureTest
[error]         bad.robot.temperature.rrd.RpnGeneratorTest
[error]         bad.robot.temperature.rrd.SecondsTest
[error]         bad.robot.temperature.rrd.ArchiveTest
[error] (temperature-machine/test:test) sbt.TestsFailedException: Tests unsuccessful
[error] Total time: 288 s, completed 31-Oct-2017 19:44:19

real    7m24.395s
user    6m25.270s
sys     0m23.100s

tobyweston, 2017-10-31:
@Cloudedhouse this looks more fundamental than SBT just flaking out. Some actual tests are failing (presumably consistently). The JAR file won't build if there are failing tests.

You could try a couple of things.

1. Update to the latest source code with `git pull` and try again. You can run `sbt test` to just compile and run the tests and get an idea if they'll at least pass.

2. Build from source on another machine then manually copy the JAR file over to `target/scala-2.12`. I do this when i'm developing/testing. You can run a command like `scp target/scala-2.12/temperature-machine-2.1.jar pi@YOUR_PI:/home/pi/code/temperature-machine/target/scala-2.12` once you've created the JAR locally.

Cloudedhouse, 2017-11-04:
Good new Toby, I started over again but with a Pi 3. Worked perfectly first time, so I think I've answered my own question, no it won't work on a Pi B.

Thanks for creating and sharing the temperature machine it's exactly what I have been looking for for a long time.

closed 2017-11-05 by tobyweston