tobyweston/temperature-machine #23: Add 6 and 12 month charts

tobyweston, 2017-10-30:
Possibly longer, how does recording temperatures for 5 years sound?

singh763173, 2017-11-06:
Cheers @tobyweston - If the data is to be kept local on the Pi, then I think 1 year is probably more suitable (given the lifespan on the device and the SD Card) but if there was a view to offloading the data to a cloud service then maybe 5 years doesn't become so much of an issue..

tobyweston, 2017-11-06:
It might be worth cataloging any investigation into external dashboards/data pumps here. I could move it into the offical docs if anything comes of it.

So far, we've mentioned:

* [Freeboard.io](https://freeboard.io/) ([source](https://github.com/Freeboard/freeboard))
* [Power BI](https://powerbi.microsoft.com/en-us/)

matty256, 2018-02-23:
I would like the ability to review data older than 1 year, for example comparing trends year over year. Ideally, logs would be stored as long as there is storage available and the UX landing page would have links to history pages that render graphs for the older data.

tobyweston, 2018-02-23:
Thanks for the feedback. I plan to add something but a basic principle of the software is a fixed database size (being on a Pi and all that). So I can see having a few years stored but not all data. Might be better to offer the option to send data elsewhere whilst recording (as discussed a little above)