tobyweston/temperature-machine #24: Export Data & Time resolution

Totoleheros, 2017-11-04:
Would be nice to have the possibility to export the data (tab.txt or csv).

Also, would be nice to have the possibility to change the timing of the temperature sampling.

tobyweston, 2017-11-05:
1. I have something I've been using to debug things. It's on a branch but https://github.com/tobyweston/temperature-machine/blob/bf81f043d5fc2cc802b1ae9ee7760101c354eb90/src/test/scala/bad/robot/temperature/ConvertJsonToCsv.scala will create a CSV from JSON. If you can wait for a bit, I'll be merging it back to master and can make it usable / add it in properly. If not, you can hack about with it to get you started.

2. I've raised #26 

3. Good idea. It's easier (for me) to track these things as an issue per request. Do you want to raise one for 1 and 3? Then close this one?

Totoleheros, 2017-11-05:
Thank you. I have open #27 and #28 issues for the points 1/ and 3/ as requested.

tobyweston, 2017-11-14:
As these have been split, closing for now.

closed 2017-11-14 by tobyweston