tobyweston/temperature-machine #25: Unable to start temperature-machine on boot

Totoleheros, 2017-11-04:
Hi,

I tried to adapt the recommendation from the documentation to start the temperature-machine on startup. In my case, the server (Pi 3, hosname=raspberrypi) has the probes, so I added this line in /etc/rc.local file (before exit 0 line):

su pi -c 'cd /home/pi/code/temperature-machine && ./start.sh raspberrypi &'

But this doesn't work. I have to launch manually the server:

sudo cd  /home/pi/code/temperature-machine
./start.sh

What am I doing wrong?

tobyweston, 2017-11-05:
What get's put in the log? Nothing I'm guessing.

I suggest you run the same command from bash and see what output you get. So

```
cd /home/pi/code/temperature-machine
./start.sh raspberrypi &
```
If that works, there's no reason I can think of why adding it into the `rc.local` file would fail. 

For reference, mine looks like this (I'm running against the `start-server.sh` script instead.
```
su pi -c 'cd /home/pi/code/temperature-machine && ./start-server.sh study bedroom1 kitchen lounge &'
```
The only difference is that the `start.sh` runs the server with the local hostname as a single argument. It will ignore your argument `raspberrypi`, whereas `start-server.sh` will run the server and take in all the arguments (`bedroom1`, `kitchen` etc in my case). 

Try both from the shell first to confirm they run then have another go!

Totoleheros, 2017-11-05:
OK, so the command from bash works perfectly well. Here is my rc.local file:
`
#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

# Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
  printf "My IP address is %s\n" "$_IP"
fi

su - pi -c 'cd /home/pi/code/temperature-machine ./start.sh raspberrypi &'
su - pi -c '/usr/bin/tightvncserver :1'
exit 0`

This file works since the tightvncserver is started as expected. I don't understand what I am doing wrong...???

tobyweston, 2017-11-14:
Looks like you're missing the `&&`:

```
su pi -c 'cd /home/pi/code/temperature-machine && ./start.sh foo &'
```

closed 2017-11-14 by tobyweston