tobyweston/temperature-machine #27: Export data into csv

Totoleheros, 2017-11-05:
As requested by tobyweston in the issue #24, I post here the point 1:

1/ Would be nice to have the possibility to export the data (tab.txt or csv).

Toby:
1/ I have something I've been using to debug things. It's on a branch but https://github.com/tobyweston/temperature-machine/blob/bf81f043d5fc2cc802b1ae9ee7760101c354eb90/src/test/scala/bad/robot/temperature/ConvertJsonToCsv.scala will create a CSV from JSON. If you can wait for a bit, I'll be merging it back to master and can make it usable / add it in properly. If not, you can hack about with it to get you started.

Totoleheros, 2017-11-05:
I could try the 'ConvertJsonToCsv.scala' but would  need some guidance to know how to run this file (I'm a newbie to Raspian and raspberry...). Thank you in advance.

tobyweston, 2018-01-02:
Moving over to #32 (didn't realise we had two of the same issue)

closed 2018-01-02 by tobyweston