tobyweston/temperature-machine #29: I can't find ~/.temperatures

Cloudedhouse, 2017-11-07:
Where is the above please?

I think it's essential to have access to it in order to make changes such as to the hostname etc

Many thanks

tobyweston, 2017-11-08:
Can you give more context what you're trying to do.

The working folder for temperature machine is `~/,temperature`. You'll find the images, source RRD etc there.

`~` means the user's home directory (it's unix shorthand). In this case, I mean the `pi` user. So when you're logged in as `pi`, you can type:

```
cd ~
ls -la
```
and if temperature-machine has ever been run, you'll see the `.temperature` folder. `cd` into that.

The long hand is ` /home/pi` so you could always type `cd  /home/pi`. Make sure you're logged in as `pi`.

closed 2017-11-08 by tobyweston

Cloudedhouse, 2017-11-08:
Ah, found it, I had to enable Root following these instructions, thanks very much.

# sudo -i
# vi /etc/ssh/sshd_config
From, PermitRootLogin without-password
To, PermitRootLogin yes
# /etc/init.d/ssh restart