tobyweston/temperature-machine #32: CSV Export

tobyweston, 2017-11-22:
From @singh763173
 
> Apologies, me again - Any chance you might be able to help me write a cron job that can query the domain.com/temperature file and copy that contents into a CSV file? 
> I'm thinking that I can create a csv with three columns, date, time, temp and then collect the data and work with that - what do you think?

tobyweston, 2017-11-22:
Couple of questions: what are you trying to achieve here? We might be able to do something longer term? Is it related to #23 ? Can you give me some more background? There's a CSV export but it only works for the last 24 hours so may not be what you want.

singh763173, 2017-11-23:
it is indeed related to #23 - I'm wanting to play with powerbi dashboards and thought this might be a quick and dirty way of getting a file with all of the data that I could consume in PBI!

tobyweston, 2017-12-31:
I've been looking at a new menu widget and the ability to export from there? What do you think @singh763173 ?

[image]

singh763173, 2018-01-01:
blimey - that's really impressive! is this in the latest release?

which graph does the export to csv feature provide?

commit 868a20e3 (2018-01-16): adding latest version of the web-app (from http://github.com/tobyweston/temperature-machine-ui) fixes #32
closed 2018-01-16 by tobyweston