tobyweston/temperature-machine #33: 7days/1month chart

andrei2186, 2017-11-26:
Hi,
Yesterday no longer displays anything on 7days/1month chart.They are empty.
 What could be the problem

Thank you.

tobyweston, 2017-11-26:
Is there anything in the log? Is it just after you started up? The charts are regenerated periodically, so the first time you start up, the charts will be empty. Refresh and it should come back.

andrei2186, 2017-11-26:
I know after start is available after 1 min. But yesterday not work all day. I will check the log and I post here.
Thank you.

singh763173, 2017-12-08:
I had the same problem but found that if I refreshed the page more than enough times, it eventually came back.

tobyweston, 2017-12-08:
has it come back @andrei2186 ?

andrei2186, 2017-12-08:
Yes. All are ok now.

andrei2186, 2017-12-08:
The problem come back.2 hours ago.

tobyweston, 2017-12-08:
Is there anything in the log? how about the browser console (in Chrome hit cmd + alt + i)?

andrei2186, 2017-12-08:
I think the problem was generated from raspberry forced shutdown. Where are the logs?

tobyweston, 2017-12-08:
Logs are in `.temperature` under your home folder.

'cat ~/.temperature/temperature-machine.log`

andrei2186, 2017-12-08:
2017-12-08 20:19:26:457 [main] INFO Starting temperature-machine (server mode)...
2017-12-08 20:19:26:547 [main] INFO RRD initialising for 'raspberrypi' (with up to 5 sensors each)...
2017-12-08 20:19:26:587 [main] INFO Ok
2017-12-08 20:19:29:192 [main] INFO Starting Discovery Server, listening for 'raspberrypi'...
2017-12-08 20:19:29:227 [temperature-machine-discovery-server-1] INFO Listening for broadcast messages...
2017-12-08 20:19:29:385 [main] INFO Monitoring sensor file(s) on 'raspberrypi'
        /sys/bus/w1/devices/28-0000070540af/w1_slave
        /sys/bus/w1/devices/28-0000076b780e/w1_slave

2017-12-08 20:19:31:201 [main] INFO HTTP Server started on http://127.0.1.1:11900

andrei2186, 2017-12-08:
[image]

tobyweston, 2018-01-04:
FYI, charts are regenerated as per the following schedule. I have no clue at the moment why that might fail but will start by reviewing the logging (as I see nothing at present in the logs). Worse case, refresh the page according to the schedule and see if they appear (at least until we can fix it)

* Every 90 seconds for the last 24 hour chart
* Every 12 hours for the last 7 days chart
* Every 24 hours for the last 30 days chart

Quaxo76, 2018-01-19:
Just to add that I see this issue a lot. The 24-hour graph is usually fine, the 7-day graph is empty most of the time, and the 1-month... I think I've only seen it display some data once or twice.

tobyweston, 2018-11-05:
Thanks for the comment, I suspect a race condition in the task startup stuff around here:

https://github.com/tobyweston/temperature-machine/blob/3263504b6cde9c7b97ae9d01254b8d09ac89166f/src/main/scala/bad/robot/temperature/task/IOs.scala#L46

but it's just a guess at this point.

A breakpoint (assuming you can reproduce) in `Graph.scala` should at least say if it's because of an empty data set. I had (vainly) assumed it was a bug in the underlying RRD before.

https://github.com/tobyweston/temperature-machine/blob/cae33c941d867094a6a792a25e16d95c3d0dec7f/src/main/scala/bad/robot/temperature/rrd/Graph.scala#L13