tobyweston/temperature-machine #35: PCB

Totoleheros, 2017-12-01:
On the http://temperature-machine.com web site, you mention the PBC that you have designed. It is pretty neat. I have found that it could be easy and cheap to order it from the JLCPCB website. Would it be possible to share the gerber file of your PCB?

tobyweston, 2017-12-01:
Defo! 

I was actually planning to do a "manufacturing run" as I've had some interest. Have you had boards from JLC before? I used Seed before and thought they were good.

I'd actually like to revise the board design and maybe incorporate a breakout or additional sensors. Any ideas?

Totoleheros, 2017-12-01:
No, just read good reviews on the web concerning this PCB maker. The board as it is right now sounds good enough for me and my needs, no further ideas to improve it.
Regards

Totoleheros, 2017-12-04:
Any chance that you will share the gerber file of your current PCB?

tobyweston, 2017-12-04:
The fritzing file is in with the source, you can generate the gerber files with fritzing...

user7743, 2017-12-06:
I would be very interested if you did a run of the PCB's. They look very need and useful for this kind of project.

tobyweston, 2017-12-06:
I've placed a little order, not sure when they'll come or the quality but happy to send a few out if you drop me a line.

user7743, 2017-12-06:
Not sure how best to contact you Toby but you can mail me at user7743@gmail.com if you wish. Id be happy to pay for the PCB ;-)

Totoleheros, 2017-12-06:
I have found the fritzing file in the source. Thank you so much!

tobyweston, 2017-12-08:
I’ve done a small “manufacturing run” (grand title). Get it touch if you’d like one (DM on Twitter? @jamanifin)

Totoleheros, 2017-12-13:
I got mine (n=10) from JLC 2 days ago. Very quick service, perfect packaging, perfect finish. And the best: for only 2€, all included (shipment too!)? I really do not know how they make money out of this business, something sounds wrong...At any rate, I'm very very happy with what I've got from them.

tobyweston, 2017-12-13:
I know! I did the same. Not sure if this is prototype only as the price goes up for more. Still, very good value, I'm going to do a revision and get another lot done.

closed 2017-12-13 by tobyweston