tobyweston/temperature-machine #36: Alert System

singh763173, 2017-12-02:
Would it be difficult to implement an alert (email?) based on an upper and lower variable. 

I.e temp has decreased below threshold. Or above threshold. 

Use case - I’ve now submerged my sensor in my tropical fish tank and want to monitor temps

tobyweston, 2017-12-02:
Good idea. Might be more flexible if we setup an alert (high and low) to work with [IFTTT](https://ifttt.com/). I think there's an "applet" for sending [emails](https://ifttt.com/email) plus you could do other interesting things...

tobyweston, 2017-12-03:
Here's my current thoughts is to have each user setup IFTTT if they want high and low alarms:

- [ ] have temperature machine to send a POST request IFTTT's to the user's [webhook](https://ifttt.com/maker_webhooks)
- [ ] each user can then setup the webhook as a trigger to use any service they want

I spiked it and had IFTTT send me an email after a post:

```
curl -X POST https://maker.ifttt.com/trigger/{event}/with/key/dGIzPjs2iHssdf9ABoj
```

So it works in principle.

singh763173, 2018-01-02:
I like it! IFTTT could make things extremely interesting! 

Going to look into Webhooks and see if I can figure out how to get it set up - I'm guessing it might be more hassle then it's worth to build into the new UI?

tobyweston, 2018-01-02:
Part of the reason for the new UI was to give options for config screens etc, so I'll see how it all goes. There's another option if I swap the library I use to read the temperatures. Maxim (the manufacturer of the temperature sensor) have their own library which offers an alerting thing. If I switch to use that, it might be super-easy to hook something up. Plus, we could prob drop the dependency on the `w1_therm` kernel module (or whatever it is).

singh763173, 2018-01-02:
What happens if someone uses a non-Maxim produced sensor? (Assuming that Maxim aren't the only players in the game...) Are their libraries universal?

tobyweston, 2018-01-02:
Only the DS18b20 sensor is supported at present. I'm sure other people have cloned the maxim sensor (I suspect I have some fakes) but as long as the kernel module works its fine. So we wouldn't be loosing out using their lib. If anything, it might support more out of the box...

https://www.maximintegrated.com/en/products/digital/one-wire/software-tools/api-java-sdk.html

johnfolsomjr, 2018-01-23:
I just wanted to also express interest in email notifications, I'm building this project as a monitor for one of our equipment room and it would be great to know when the A/C decides to die over the weekend....

tobyweston, 2018-01-23:
👍