tobyweston/temperature-machine #38: Changed Hostname and Device has stopped logging

singh763173, 2017-12-30:
Hey Toby,

Wondering if you could help me out - I changed the hostname on my Pi last night and now it has stopped logging. The page is still available on port 11900 and when i cat the w1_slave file for the sensor, I'm getting a reading.

I'm guessing the hostname is hardcoded somewhere and needs to be updated? Any idea where?

Cheers

singh763173, 2017-12-30:
`cat /home/pi/.temperature/temperature-machine.log 
Error: Unable to access jarfile target/scala-2.12/temperature-machine-2.1.jar
An error occurred executed a scheduled task (GenerateGraph(24 hours)) Unknown datasource name: tempmonitor-sensor-1
An error occurred executed a scheduled task (JsonExport(24 hours)) Unknown datasource name: tempmonitor-sensor-1`

Old hostname - TempMonitor01
New hostname - tempmonitor1

tobyweston, 2017-12-30:
Hey, check your .sh files (start.sh, start-client.sh and start-server.sh) and the boot setup in /etc/dc.local. There may be mention of your host name. Let me know how you get on.

tobyweston, 2018-01-01:
@singh763173 is this working now? I'll close if so

singh763173, 2018-01-02:
I think so - just the issue that I posted in the other thread!

Thanks for your help with this one!

closed 2018-01-02 by singh763173