tobyweston/temperature-machine #39: Issues building new device

singh763173, 2017-12-30:
Me again (this isn't related to any other issues)

I'm currently building a new Pi as i want to log temps elsewhere but getting the following issues when running sbt assembly

`[info] Checking every *.class/*.jar file's SHA-1.
[info] Merging files...
[warn] Merging 'META-INF/MANIFEST.MF' with strategy 'discard'
[warn] Strategy 'discard' was applied to a file
[info] SHA-1: 825dc9a8fcfcf8e80307c9fbec336c195183b74d
[info] Packaging /home/pi/code/temperature-machine/site/target/scala-2.10/site-assembly-0.1-SNAPSHOT.jar ...
[info] Done packaging.
[info] Passed: Total 109, Failed 0, Errors 0, Passed 109
[info] Checking every *.class/*.jar file's SHA-1.
[info] Merging files...
[warn] Merging 'META-INF/NOTICE' with strategy 'rename'
[warn] Merging 'META-INF/LICENSE.txt' with strategy 'rename'
[warn] Merging 'META-INF/LICENSE' with strategy 'rename'
[warn] Merging 'META-INF/DEPENDENCIES' with strategy 'discard'
[warn] Merging 'META-INF/MANIFEST.MF' with strategy 'discard'
[warn] Merging 'META-INF/maven/com.twitter/hpack/pom.properties' with strategy 'discard'
[warn] Merging 'META-INF/maven/com.twitter/hpack/pom.xml' with strategy 'discard'
[warn] Merging 'META-INF/maven/org.apache.logging.log4j/log4j-api/pom.properties' with strategy 'discard'
[warn] Merging 'META-INF/maven/org.apache.logging.log4j/log4j-api/pom.xml' with strategy 'discard'
[warn] Merging 'META-INF/maven/org.apache.logging.log4j/log4j-core/pom.properties' with strategy 'discard'
[warn] Merging 'META-INF/maven/org.apache.logging.log4j/log4j-core/pom.xml' with strategy 'discard'
[warn] Merging 'META-INF/maven/org.apache.logging.log4j/log4j-slf4j-impl/pom.properties' with strategy 'discard'
[warn] Merging 'META-INF/maven/org.apache.logging.log4j/log4j-slf4j-impl/pom.xml' with strategy 'discard'
[warn] Merging 'META-INF/maven/org.eclipse.jetty.alpn/alpn-api/pom.properties' with strategy 'discard'
[warn] Merging 'META-INF/maven/org.eclipse.jetty.alpn/alpn-api/pom.xml' with strategy 'discard'
[warn] Merging 'META-INF/maven/org.rrd4j/rrd4j/pom.properties' with strategy 'discard'
[warn] Merging 'META-INF/maven/org.rrd4j/rrd4j/pom.xml' with strategy 'discard'
[warn] Merging 'META-INF/maven/org.slf4j/slf4j-api/pom.properties' with strategy 'discard'
[warn] Merging 'META-INF/maven/org.slf4j/slf4j-api/pom.xml' with strategy 'discard'
[error] 1 error was encountered during merge
java.lang.RuntimeException: deduplicate: different file contents found in the following:
/root/.ivy2/cache/org.apache.logging.log4j/log4j-api/jars/log4j-api-2.10.0.jar:module-info.class
/root/.ivy2/cache/org.slf4j/slf4j-api/jars/slf4j-api-1.8.0-alpha2.jar:module-info.class
	at sbtassembly.Assembly$.applyStrategies(Assembly.scala:140)
	at sbtassembly.Assembly$.x$1$lzycompute$1(Assembly.scala:25)
	at sbtassembly.Assembly$.x$1$1(Assembly.scala:23)
	at sbtassembly.Assembly$.stratMapping$lzycompute$1(Assembly.scala:23)
	at sbtassembly.Assembly$.stratMapping$1(Assembly.scala:23)
	at sbtassembly.Assembly$.inputs$lzycompute$1(Assembly.scala:67)
	at sbtassembly.Assembly$.inputs$1(Assembly.scala:57)
	at sbtassembly.Assembly$.apply(Assembly.scala:83)
	at sbtassembly.Assembly$$anonfun$assemblyTask$1.apply(Assembly.scala:240)
	at sbtassembly.Assembly$$anonfun$assemblyTask$1.apply(Assembly.scala:237)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
	at sbt.Execute.work(Execute.scala:237)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
	at sbt.ConcurrentRestrictions$$anon$4$$anonfun$1.apply(ConcurrentRestrictions.scala:159)
	at sbt.CompletionService$$anon$2.call(CompletionService.scala:28)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
[error] (temperature-machine/*:assembly) deduplicate: different file contents found in the following:
[error] /root/.ivy2/cache/org.apache.logging.log4j/log4j-api/jars/log4j-api-2.10.0.jar:module-info.class
[error] /root/.ivy2/cache/org.slf4j/slf4j-api/jars/slf4j-api-1.8.0-alpha2.jar:module-info.class
[error] Total time: 1092 s, completed 30-Dec-2017 13:35:15
pi@tempmonitor2:~/code/temperature-machine $ 
`

tobyweston, 2017-12-30:
Standard procedure is to update with ‘git pull’ and try again. There’s a fair bit of development going on so you may have caught an unlucky build. 

I’m working on a major version so keep and eye out. I’ll aim to package it slightly differently to avoid people having to rebuild (which is always annoying on a pi).

The other thing to try is ‘sbt clean’ before you try to build. You can also build in stages; run ‘sbt test:compile’ before you run ‘assembly’. The pi (zero especially) struggles a bit so it seems to help.

Ps the new version will have an export feature btw

singh763173, 2017-12-31:
Great news with regards to the export feature!

I just tried to update my current monitor and had the exact same issue - will try to run sbt clean and then re-run the assembly to see how that goes. Failing that, I may have to to revert to the previous commit (unless the next release isn't far off?)

Hope you had a great Christmas and Happy New Year in advance!

tobyweston, 2017-12-31:
No ETA on the release but I could have a go at a preview release if I get the chance. Will let you know.

Cheers and HNY too!

tobyweston, 2018-01-01:
I've been able to reproduce on a different machine, so it looks like its a genuine problem that wasn't showing up on my main machine. I'll report back with a fix.

```
[warn] Merging 'META-INF/maven/org.slf4j/slf4j-api/pom.xml' with strategy 'discard'
[error] 1 error was encountered during merge
java.lang.RuntimeException: deduplicate: different file contents found in the following:
/Users/toby/.ivy2/cache/org.apache.logging.log4j/log4j-api/jars/log4j-api-2.10.0.jar:module-info.class
/Users/toby/.ivy2/cache/org.slf4j/slf4j-api/jars/slf4j-api-1.8.0-alpha2.jar:module-info.class
	at sbtassembly.Assembly$.applyStrategies(Assembly.scala:140)
	at sbtassembly.Assembly$.x$1$lzycompute$1(Assembly.scala:25)
	at sbtassembly.Assembly$.x$1$1(Assembly.scala:23)
	at sbtassembly.Assembly$.stratMapping$lzycompute$1(Assembly.scala:23)
	at sbtassembly.Assembly$.stratMapping$1(Assembly.scala:23)
	at sbtassembly.Assembly$.inputs$lzycompute$1(Assembly.scala:67)
	at sbtassembly.Assembly$.inputs$1(Assembly.scala:57)
	at sbtassembly.Assembly$.apply(Assembly.scala:83)
	at sbtassembly.Assembly$$anonfun$assemblyTask$1.apply(Assembly.scala:240)
	at sbtassembly.Assembly$$anonfun$assemblyTask$1.apply(Assembly.scala:237)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
	at sbt.Execute.work(Execute.scala:237)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
	at sbt.ConcurrentRestrictions$$anon$4$$anonfun$1.apply(ConcurrentRestrictions.scala:159)
	at sbt.CompletionService$$anon$2.call(CompletionService.scala:28)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
[error] (temperature-machine/*:assembly) deduplicate: different file contents found in the following:
[error] /Users/toby/.ivy2/cache/org.apache.logging.log4j/log4j-api/jars/log4j-api-2.10.0.jar:module-info.class
[error] /Users/toby/.ivy2/cache/org.slf4j/slf4j-api/jars/slf4j-api-1.8.0-alpha2.jar:module-info.class
[error] Total time: 32 s, completed 01-Jan-2018 17:22:15
$ 
```

tobyweston, 2018-01-01:
I think this was caused by a duplicate file coming from SLF4J, removing that dependence seems to fix it but then there's no default binding causing:

```
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
```
I think the lesser of two evils is to change the merge strategy for `assembly` (https://github.com/sbt/sbt-assembly#merge-strategy). No idea to what though...

For reference, the dependencies are below.

```
> whatDependsOn org.slf4j slf4j-api 1.8.0-alpha2
[info] 
[info] org.slf4j:slf4j-api:1.8.0-alpha2
[info]   +-org.apache.logging.log4j:log4j-slf4j-impl:2.10.0
[info]   | +-bad.robot:temperature-machine_2.12:2.1 [S]
[info]   | 
[info]   +-org.log4s:log4s_2.12:1.3.6 [S]
[info]     +-org.http4s:blaze-core_2.12:0.12.9 [S]
[info]     | +-org.http4s:blaze-http_2.12:0.12.9 [S]
[info]     |   +-org.http4s:http4s-blaze-core_2.12:0.16.5a [S]
[info]     |     +-org.http4s:http4s-blaze-client_2.12:0.16.5a [S]
[info]     |     | +-bad.robot:temperature-machine_2.12:2.1 [S]
[info]     |     | 
[info]     |     +-org.http4s:http4s-blaze-server_2.12:0.16.5a [S]
[info]     |       +-bad.robot:temperature-machine_2.12:2.1 [S]
[info]     |       
[info]     +-org.http4s:blaze-http_2.12:0.12.9 [S]
[info]     | +-org.http4s:http4s-blaze-core_2.12:0.16.5a [S]
[info]     |   +-org.http4s:http4s-blaze-client_2.12:0.16.5a [S]
[info]     |   | +-bad.robot:temperature-machine_2.12:2.1 [S]
[info]     |   | 
[info]     |   +-org.http4s:http4s-blaze-server_2.12:0.16.5a [S]
[info]     |     +-bad.robot:temperature-machine_2.12:2.1 [S]
[info]     |     
[info]     +-org.http4s:http4s-core_2.12:0.16.5a [S]
[info]       +-org.http4s:http4s-argonaut_2.12:0.16.5a [S]
[info]       | +-bad.robot:temperature-machine_2.12:2.1 [S]
[info]       | 
[info]       +-org.http4s:http4s-blaze-core_2.12:0.16.5a [S]
[info]       | +-org.http4s:http4s-blaze-client_2.12:0.16.5a [S]
[info]       | | +-bad.robot:temperature-machine_2.12:2.1 [S]
[info]       | | 
[info]       | +-org.http4s:http4s-blaze-server_2.12:0.16.5a [S]
[info]       |   +-bad.robot:temperature-machine_2.12:2.1 [S]
[info]       |   
[info]       +-org.http4s:http4s-client_2.12:0.16.5a [S]
[info]       | +-org.http4s:http4s-blaze-client_2.12:0.16.5a [S]
[info]       |   +-bad.robot:temperature-machine_2.12:2.1 [S]
[info]       |   
[info]       +-org.http4s:http4s-dsl_2.12:0.16.5a [S]
[info]       | +-bad.robot:temperature-machine_2.12:2.1 [S]
[info]       | 
[info]       +-org.http4s:http4s-jawn_2.12:0.16.5a [S]
[info]       | +-org.http4s:http4s-argonaut_2.12:0.16.5a [S]
[info]       |   +-bad.robot:temperature-machine_2.12:2.1 [S]
[info]       |   
[info]       +-org.http4s:http4s-server_2.12:0.16.5a [S]
[info]         +-org.http4s:http4s-blaze-server_2.12:0.16.5a [S]
[info]           +-bad.robot:temperature-machine_2.12:2.1 [S]
[info]           
[success] Total time: 0 s, completed 01-Jan-2018 17:43:47
```

tobyweston, 2018-01-01:
10d00a8 fixes this (hopefully), have a go and report back!

singh763173, 2018-01-01:
@tobyweston completed successfully! looks like that did the trick - however I've now got a new problem - not sure if it's related. 

sbt assembly completes successfully, but when running start.sh I'm not able to load the site. shall i raise a new issue?

singh763173, 2018-01-01:
strange behaviour - 
tempmonitor1 (the original one) now presents the WebUI but i get the following error

`TypeError: n.data.map is not a function. (In 'n.data.map(function(e,n){return Object.assign({},e,{fill:!1,radius:0,spanGaps:!1,borderColor:t[n],backgroundColor:t[n]})})', 'n.data.map' is undefined)`

tempmonitor2 (the new device) won't load the WebUI at all, even though sbt assembly completed successfully

tobyweston, 2018-01-01:
The `n.data` thing is usually because the front end (UI) has started but can't speak to the backend for some reason. Is it just on the chart or on the temp values too? Screenshot?

For tempmonitor2: don't forget you have just one server and many clients. So say tempmonitor1 is the server and it's address is 10.0.1.10 and tempmonitor2 is your new client node at 10.0.1.20. You only start one as a server (with `start-server.sh`), the other you start as a client (with `start-client.sh`). The website is served from 10.0.1.10 in our example (the server), so after the second machine (10.0.1.20) has started up (check the logs that it started ok), you'd just go to the usual site (10.0.1.10:11900) and you'd see both sets of temperatures. Don't try and go to 10.0.1.20:11900 (the client) as there's no website there!

Not sure if that was obvious or not, apologies if you know all that already and thanks for bearing with me. It should be covered in the website docs. Give me a shout if that's not clear etc.

singh763173, 2018-01-01:
tempmonitor1 - only the graph - the value above is coming out fine - screen shot below

[image]

tempmonitor2 - update - found that sbt was permission with root which was causing some issues so chown and chgrp to pi - now when i go to run sbt assembly it throws a java error! unless i sudo sbt assembly which completes but still doesn't show the site
completely understood with regards to client/server setup - i wanted to run them both as distinct servers, serving there own sensor starts individually and once it's proving to be working then run the server on 1 and client on 2. i assumed that having two servers wouldn't be an issue?

singh763173, 2018-01-01:
tempmonitor2 java error - incase it's useful

`pi@tempmonitor2:~/code/temperature-machine $ sbt assembly
java.io.IOException: Permission denied
	at java.io.UnixFileSystem.createFileExclusively(Native Method)
	at java.io.File.createNewFile(File.java:1012)
	at scala.reflect.io.AbstractFile.fileOrSubdirectoryNamed(AbstractFile.scala:235)
	at scala.reflect.io.AbstractFile.fileNamed(AbstractFile.scala:246)
	at scala.tools.nsc.backend.jvm.BytecodeWriters$class.getFile(BytecodeWriters.scala:34)
	at scala.tools.nsc.backend.jvm.BytecodeWriters$class.scala$tools$nsc$backend$jvm$BytecodeWriters$$getFile(BytecodeWriters.scala:37)
	at scala.tools.nsc.backend.jvm.BytecodeWriters$ClassBytecodeWriter$class.writeClass(BytecodeWriters.scala:89)
	at scala.tools.nsc.backend.jvm.GenASM$AsmPhase$$anon$4.writeClass(GenASM.scala:67)
	at scala.tools.nsc.backend.jvm.GenASM$JBuilder.writeIfNotTooBig(GenASM.scala:459)
	at scala.tools.nsc.backend.jvm.GenASM$JMirrorBuilder.genMirrorClass(GenASM.scala:2980)
	at scala.tools.nsc.backend.jvm.GenASM$AsmPhase.run(GenASM.scala:114)
	at sbt.compiler.Eval$$anonfun$compile$1$1.apply$mcV$sp(Eval.scala:177)
	at sbt.compiler.Eval$$anonfun$compile$1$1.apply(Eval.scala:177)
	at sbt.compiler.Eval$$anonfun$compile$1$1.apply(Eval.scala:177)
	at scala.reflect.internal.SymbolTable.atPhase(SymbolTable.scala:207)
	at sbt.compiler.Eval.compile$1(Eval.scala:177)
	at sbt.compiler.Eval.compileAndLoad(Eval.scala:182)
	at sbt.compiler.Eval.evalCommon(Eval.scala:152)
	at sbt.compiler.Eval.eval(Eval.scala:96)
	at sbt.EvaluateConfigurations$.evaluateDslEntry(EvaluateConfigurations.scala:177)
	at sbt.EvaluateConfigurations$$anonfun$9.apply(EvaluateConfigurations.scala:117)
	at sbt.EvaluateConfigurations$$anonfun$9.apply(EvaluateConfigurations.scala:115)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:244)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:244)
	at scala.collection.immutable.List.foreach(List.scala:318)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:244)
	at scala.collection.AbstractTraversable.map(Traversable.scala:105)
	at sbt.EvaluateConfigurations$.evaluateSbtFile(EvaluateConfigurations.scala:115)
	at sbt.Load$.sbt$Load$$loadSettingsFile$1(Load.scala:775)
	at sbt.Load$$anonfun$sbt$Load$$memoLoadSettingsFile$1$1.apply(Load.scala:781)
	at sbt.Load$$anonfun$sbt$Load$$memoLoadSettingsFile$1$1.apply(Load.scala:780)
	at scala.collection.MapLike$class.getOrElse(MapLike.scala:128)
	at scala.collection.AbstractMap.getOrElse(Map.scala:58)
	at sbt.Load$.sbt$Load$$memoLoadSettingsFile$1(Load.scala:780)
	at sbt.Load$$anonfun$loadFiles$1$2.apply(Load.scala:788)
	at sbt.Load$$anonfun$loadFiles$1$2.apply(Load.scala:788)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:244)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:244)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:47)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:244)
	at scala.collection.AbstractTraversable.map(Traversable.scala:105)
	at sbt.Load$.loadFiles$1(Load.scala:788)
	at sbt.Load$.discoverProjects(Load.scala:799)
	at sbt.Load$.discover$1(Load.scala:585)
	at sbt.Load$.sbt$Load$$loadTransitive(Load.scala:633)
	at sbt.Load$$anonfun$loadUnit$1.sbt$Load$$anonfun$$loadProjects$1(Load.scala:482)
	at sbt.Load$$anonfun$loadUnit$1$$anonfun$40.apply(Load.scala:485)
	at sbt.Load$$anonfun$loadUnit$1$$anonfun$40.apply(Load.scala:485)
	at sbt.Load$.timed(Load.scala:1025)
	at sbt.Load$$anonfun$loadUnit$1.apply(Load.scala:485)
	at sbt.Load$$anonfun$loadUnit$1.apply(Load.scala:459)
	at sbt.Load$.timed(Load.scala:1025)
	at sbt.Load$.loadUnit(Load.scala:459)
	at sbt.Load$$anonfun$25$$anonfun$apply$14.apply(Load.scala:311)
	at sbt.Load$$anonfun$25$$anonfun$apply$14.apply(Load.scala:310)
	at sbt.BuildLoader$$anonfun$componentLoader$1$$anonfun$apply$4$$anonfun$apply$5$$anonfun$apply$6.apply(BuildLoader.scala:91)
	at sbt.BuildLoader$$anonfun$componentLoader$1$$anonfun$apply$4$$anonfun$apply$5$$anonfun$apply$6.apply(BuildLoader.scala:90)
	at sbt.BuildLoader.apply(BuildLoader.scala:140)
	at sbt.Load$.loadAll(Load.scala:365)
	at sbt.Load$.loadURI(Load.scala:320)
	at sbt.Load$.load(Load.scala:316)
	at sbt.Load$.load(Load.scala:305)
	at sbt.Load$$anonfun$4.apply(Load.scala:146)
	at sbt.Load$$anonfun$4.apply(Load.scala:146)
	at sbt.Load$.timed(Load.scala:1025)
[... 41 lines cut ...]
	at sbt.MainLoop$.run(MainLoop.scala:89)
	at sbt.MainLoop$$anonfun$runWithNewLog$1.apply(MainLoop.scala:68)
	at sbt.MainLoop$$anonfun$runWithNewLog$1.apply(MainLoop.scala:63)
	at sbt.Using.apply(Using.scala:24)
	at sbt.MainLoop$.runWithNewLog(MainLoop.scala:63)
	at sbt.MainLoop$.runAndClearLast(MainLoop.scala:46)
	at sbt.MainLoop$.runLoggedLoop(MainLoop.scala:30)
	at sbt.MainLoop$.runLogged(MainLoop.scala:22)
	at sbt.StandardMain$.runManaged(Main.scala:57)
	at sbt.xMain.run(Main.scala:29)
	at xsbt.boot.Launch$$anonfun$run$1.apply(Launch.scala:109)
	at xsbt.boot.Launch$.withContextLoader(Launch.scala:128)
	at xsbt.boot.Launch$.run(Launch.scala:109)
	at xsbt.boot.Launch$$anonfun$apply$1.apply(Launch.scala:35)
	at xsbt.boot.Launch$.launch(Launch.scala:117)
	at xsbt.boot.Launch$.apply(Launch.scala:18)
	at xsbt.boot.Boot$.runImpl(Boot.scala:41)
	at xsbt.boot.Boot$.main(Boot.scala:17)
	at xsbt.boot.Boot.main(Boot.scala)
[error] java.io.IOException: Permission denied
[error] Use 'last' for the full log.
Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? 
`

clearly a permission issue but not sure what files might be incorrectly permissioned

tobyweston, 2018-01-01:
Running two servers will be a problem! The server broadcasts it's address so that clients can self discover where to send their data. If two servers are on the same network, they will probably fight for where the data is sent. Having said that, I think you might get away with it if you just have two servers and no clients (as the server's send data to themselves, circumventing the broadcast/self discover mechanism. I think! I'd never considered that scenario.

RE the stacktrace: it looks like you still have permission problems with something running as root. Maybe a folder is owned by root under `code`? If sbt once ran as root from that folder and it needed to create folders, they'd be owned by root. Perhaps `sudo rm -rf target` and try again?

singh763173, 2018-01-01:
where would i run the remove command? within /usr/local/bin? if so, i've killed both sbt and the jar file and recreated and tried again to no avail. when i ran it again, it downloaded all the required files again but i guess it's still referencing the files downloaded as root but not sure where it downloads them to

singh763173, 2018-01-01:
nvm - i just figured out what you meant... it's in the code folder... apologies

singh763173, 2018-01-01:
deleted the target folder and tried again and same permission error

tobyweston, 2018-01-02:
Yep, I mean under `code`. It's prob worth trying to clear down and start from fresh at this point if you can't find the folders.

There's a hidden folder that sbt uses under `~`, something like `.sbt` and `.ivy`. You could try deleting those too. Check under the `root` user's home too (not just `pi`). Try deleting `code` completly and redownloading etc?

tobyweston, 2018-01-02:
You could try something like:

```
cd ~/code
find . -user root
```

to list things owned by `root`. Nothing should be there.

You could then try it from the `pi` user's home folder.

The syntax might be slightly off so test it first 😄

singh763173, 2018-01-02:
cheers Toby - deleting the code dir and cloning again appears to have done the trick and allowed me to compile. Not sure what happened there.

However - Still not presented with the webui.

I've noticed that i have a dodgy sensor and will have to solder a new one - is it safe to assume that the webui should still be presented, regardless of whether a sensor is present or not?

tobyweston, 2018-01-02:
No, things wont startup without sensors. You should see something in the log though saying that.

btw, if you're interested in multiple sensors, I have some spare add-on boards available for free. http://temperature-machine.com/docs/add-on_board.html

singh763173, 2018-01-02:
I’d be absolutely interested! That would be simplify my setup massively!

If you have a 2 or 3, I’d be more than happy to reimburse you?

tobyweston, 2018-01-02:
Can you share you email privately somehow (twitter DM?) and I'll arrange postage. Although I need to solder them up so might take me a week or two!

singh763173, 2018-01-03:
Wasn't sure if I found the right Toby Weston on twitter so dropped you a LinkedIn InMail instead! A little unorthodox... but it works i guess lol.

No worries with regards to time - take as long as you need!

Thank you!

tobyweston, 2018-01-03:
Message received :email:

singh763173, 2018-01-03:
Latest git pull error

`2018-01-03 16:53:42:186 [Thread-7] ERROR An error occurred listening for serv                                                                                er discovery messages. Remote machines may not be able to publish their data                                                                                 to the server. Address already in use
2018-01-03 16:53:42:251 [Thread-7] INFO Discovery server shutdown
[info] DiscoveryServerTest
[info]
[info] When the discovery server is started
[error]   x Server will respond with it's address once pinged
[error]    -\/(Timeout(socket timed out after 30 seconds ms)) is not \/- (Dis                                                                                coveryServerTest.scala:25)
[info]
[error]   x Server response when unknown message is received
[error]    -\/(Timeout(socket timed out after 30 seconds ms)) is not \/- (Dis                                                                                coveryServerTest.scala:40)
`

tobyweston, 2018-01-03:
Port clash. Both servers are trying to start up on port 11900. I should have predicted that.

singh763173, 2018-01-04:
Ah - that would make sense!

Still having issues with the graph not showing as per my screenshot above :(

tobyweston, 2018-01-04:
The chart thing is intermittent, I've seen it once or twice. Its actually happening for me on one of the three graphs right now. I think I'll close this ticket and raise another for that issue (although I wont get to it for a while as the work around seems to be give it time to regenerate and then refresh the page). Images regenerate as follows:

* Every 90 seconds for the last 24 hour chart
* Every 12 hours for the last 7 days chart
* Every 24 hours for the last 30 days chart

commit 10d00a81 (2018-01-01): fixes #39 although I'm not sure which merge stragegy is the best in this case.
closed 2018-01-04 by tobyweston