tobyweston/temperature-machine #40: Long term graphs title

Quaxo76, 2018-01-17:
I just updated to the latest available version. I noticed that when I click on one of the long range graphs, and it enlarges, at the top left of the screen I get the "title" of that graph (i.e. 24-hour or 1 week). If then I change to another graph using the left/right arrows on screen, the graph does change, but the title does not. It should change to constantly reflect what graph is being shown...

Cristian

tobyweston, 2018-01-18:
Good spot. Will investigate

closed 2018-01-18 by tobyweston

tobyweston, 2018-01-18:
Fixed by https://github.com/tobyweston/temperature-machine-ui/commit/7ea30cfc4bf1db2ed18edd616b87d3b58b6a19e2

Quaxo76, 2018-01-20:
Thank you, it works now. Could I also suggest to also add the descriptor to the actual graph? Where it says "temperature" on the top center of the graph, it could say "temperature 24 hours", "temperature 7 days" etc. This would be good especially since I'm storing the past graphs on the sd...

tobyweston, 2018-02-24:
Fixed the second part and labeled the charts better.

commit 9d28b720 (2018-01-18): adding latest version of the web-app (from http://github.com/tobyweston/temperature-machine-ui) fixes #40 lightbox title in graphs not updating