tobyweston/temperature-machine #41: Unwanted spurious spikes

Quaxo76, 2018-01-17:
My server has been behaving well, with no drops and no spikes. But the client, which has two sensors attached, sometimes shows large spikes, which are annoying especially because they mess up the autoscale of the y axis, and so I loose detail.
What is weird is that both sensors show the same spike, and it appears to be on only one frame. Could this be a communication error? It's hard to believe that both sensors always show the same spike at exactly the same moment...
[image]

Quaxo76, 2018-01-18:
Just realized this is a duplicate of #9 . I tried to apply the proposed solution by modifying my startup command to
`su pi -c 'cd /home/pi/code/temperature-machine && ./start-client.sh -Davoid.spikes=30 &'`
and rebooted. I'll see if it helps...

Cristian

tobyweston, 2018-01-18:
Great, make sure you get the latest version for that feature.

This has been a huge pain for me as its hard to reproduce and I get it all the time across machines. I really have no idea at this point as the smoothing feature doesn't work for me (although its tested so works "in theory"). I've started to suspect electrical interference!

The new download CSV feature is partly there to help me figure out why the spikes aren't getting smoothed. Perhaps you could grab some images and CSV and attach to #9 and we'll see if it helps diagnose?

Closing this issue.

closed 2018-01-18 by tobyweston