tobyweston/temperature-machine #42: A few suggestions / feature requests

Quaxo76, 2018-01-17:
Hello,
after successfully using this system for a few days I have some suggestions, which are not bug reports but more like "feature requests"...

1) It would be nice to have the current version printed somewhere on the web page... just for reference, and to know if an update worked fine. Maybe near the copyright message on the bottom? Or as an "about" entry on the side menu...

2) I noticed that the numeric temperature value updates automatically, but the graph doesn't. It would be nice to have it update automatically - in real time or even just every x minutes. That would allow us to leave a browser window open, with a constantly up-to-date graph being shown.

3) The main graph and the 24-hour small graph are very smooth. But the 1-week and 1-month are "stepped". I suppose a reading is taken every so often, and then replicated for several pixels, until the new reading is taken. But this means that the graphs are quite "messy" and a bit meaningless if the temperature oscillates regularly, as is the case where a thermostat keeps the temp within a certain range. If for example 2 or 3 readings happen to be taken when the temperature is at the max value, then the following reading happens when the temp is at the minimum, then another when it's at max again... the graph would show a long "high temp" phase, followed by a low peak. This is not very representative. It would be nice if every "horizontal segment" could be made to show the average value of all the readings of the respective time interval. Even better would be if every pixel had its own value, to avoid the "stepped" look...

4) It would be nice to have an option to configure the y-axis range (especially on the main graph, but also on the smaller ones); maybe an option to leave it to automatic, or manually set a range with minimum and/or maximum temperature... The same goes for the "green zones" on the long term graphs. Ideally these should be configurable from the web page, but a config file would do as well. This would also avoid the "compressing" effect of spurious spikes...

Thank you again for a wonderful piece of software!
Cristian

tobyweston, 2018-01-18:
Thanks for the thoughts.

1. #22 
2. This is deliberate, my use case is that I leave the browser open then might come back a few days later and still want to see what it has been plotting. I've laboured this decision though and recognise the need for it. I may well make it switchable. 
3. Can you share some examples? I suspect this is the nature of the aggregated views for > 24 hr charts. It has to drop data. Have a look at RRD and how it handles archive data.
4. Can do
5. Good idea, if you share you scripts I'll at least add them to the docs site. Raise an issue and we can discuss and evolve the idea there.

Can you create issues for each please? One issue: one feature. It makes it easier for me to manage and code against single issues then close and track each, one at a time. 

Thanks again

tobyweston, 2018-01-18:
Think these are all covered off in #46, #47, #45 and #44 (thanks for adding those), so I'll close this one.

closed 2018-01-18 by tobyweston