tobyweston/temperature-machine #45: Automatic or fixed Y-axis (temperature) range

Quaxo76, 2018-01-18:
It would be nice to have an option to configure the y-axis range (especially on the main graph, but also on the smaller ones); maybe an option to leave it to automatic, or manually set a range with minimum and/or maximum temperature... The same goes for the "green zones" on the long term graphs. Ideally these should be configurable from the web page, but a config file would do as well. This would also avoid the "compressing" effect of spurious spikes...

Cristian

tobyweston, 2018-01-18:
I'm going to create some form of config file for various requests then allow this to be editable in app, so that will fit in nicely. The compressing effect #9 needs a proper fix and is WIP!

Quaxo76, 2018-01-20:
Adding one more idea. Once we have selectable auto- or manual- ranging, what about adding checkboxes to choose which sensor's data gets drawn on the main graph? This way, keeping auto-ranging on, we could select, for example, just the incubator's sensor, so that the range is narrow and we can fully appreciate small temperature changes; or we could select for example an indoor sensor and one from outdoors, to see how external temperature affects the inside environment, or the working of the furnace... :)
Smaller graphs could keep recording all sensors.

tobyweston, 2018-01-20:
If you click the sensor name on the main chart, it already toggles the display of that line. Is that what you mean?

Quaxo76, 2018-01-20:
Ah yes, that's exactly it. I didn't know you could do that. Perhaps a checkbox next to the sensor name would be more intuitive? Anyway, even the autoscaling works perfectly, that's exactly what I meant. Thanks.

Quaxo76, 2018-01-20:
Toby, I see you closed this, but actually the issue was about the addition of a selectable auto-range or manual-range for the graph... then I hijacked the thread (my bad) with the additional idea of selectable sensors. My bad. But if I'm not mistaken, the original issue still stands, so shouldn't this still be open?