tobyweston/temperature-machine #47: Smoothing the weekly and monthly graphs

Quaxo76, 2018-01-18:
The main graph and the 24-hour small graph are very smooth. But the 1-week and 1-month are "stepped". I suppose a reading is taken every so often, and then replicated for several pixels, until the new reading is taken. But this means that the graphs are quite "messy" and a bit meaningless if the temperature oscillates regularly, as is the case where a thermostat keeps the temp within a certain range. If for example 2 or 3 readings happen to be taken when the temperature is at the max value, then the following reading happens when the temp is at the minimum, then another when it's at max again... the graph would show a long "high temp" phase, followed by a low peak. This is not very representative. It would be nice if every "horizontal segment" could be made to show the average value of all the readings of the respective time interval. Even better would be if every pixel had its own value, to avoid the "stepped" look...
I assume that the code that builds the graphs, just reads the relevant values from the RRD (one every x values). It could be better if the code could read ALL the values between a graph point and the previous one, and then display the AVERAGE of all those values. I think this would smooth the graph, avoiding those kind of "moire' artifacts"...

Cristian

tobyweston, 2018-01-18:
RRD is dealing with all of this, it has the capability for all this (I think) and just depends on how you configure it's archives. So for the weekly archive, it's setup like this (in [RRD speak](http://oss.oetiker.ch/rrdtool/doc/rrdcreate.en.html)):

   new ArcDef(AVERAGE, 0.5, 120, 168) 

I'll have to refresh my memory on RRD and perhaps we can tweak the aggregating definition above and see how it charts.

Quaxo76, 2018-01-19:
OK, so if I understand correctly, when it's time to update the graph with a new segment, the software reads from the RRD all the previous values until the previous update, averages them, and draws a short horizontal segment with that average value. Is this correct?
In this case, to avoid the "stepped" look, it would be enough to force more frequent updates, so that each pixel in the graph has its own value (which is in turn the average of a bunch of readings in the RRD). Would it add too much load on the CPU to recreate the graphs for each pixel? For the 7-day graph it's about 120 pixels per day, so it comes to about one pixel every 12 minutes. Much less than that for the monthly graph, of course...
Is this at all possible, or did I completely misunderstand how it works?