tobyweston/temperature-machine #55: LogParserTest fails when using different timezones

Quaxo76, 2018-02-06:
I live in Italy and have set my local timezone accordingly. I have to do that, because otherwise, the graphs will show the wrong time.
But with the changed timezone, the LogParserTest always fails:

[info] LogParserTest
[info] 
[error] x Example inputs
[error]  LogMessage(2017-10-31T09:40:05.976Z,[main],INFO,Starting temperature-machine (server mode)...) != LogMessage(2017-10-31T10:40:05.976Z,[main],INFO,Starting temperature-machine (server mode)...) (LogParserTest.scala:97)
[error] Actual:   ...0-31T[09]:4...Z,[main],INFO...
[error] Expected: ...0-31T[10]:4...Z,[main],INFO...

I'm forced to revert the timezone to UTC to compile, then change it back (which is annoying) or to skip the tests altogether (which defeats the point of having those tests).

tobyweston, 2018-02-07:
Fix up the tests to new up a specific instant rather than parse (parsing will invoke timezone voodoo)
```
  "Example inputs" >> {
     assert(
-      "2017-10-31 10:40:05:976 [main] INFO Starting temperature-machine (server mode)...", 
-      LogMessage(Instant.parse("2017-10-31T10:40:05.976Z"), "[main]", "INFO", "Starting temperature-machine (server mode)...")
-    )
-    assert(
-      "2017-10-31 10:40:06:220 [main] INFO Ok", 
-      LogMessage(Instant.parse("2017-10-31T10:40:06.220Z"), "[main]", "INFO", "Ok")
+      "2017-10-31 10:40:05:001 [main] INFO Starting temperature-machine (server mode)...",
+      expected = LogMessage(Instant.ofEpochMilli(1509446405001L), "[main]", "INFO", "Starting temperature-machine (server mode)...")
     )

```

tobyweston, 2018-02-07:
see #56 for wider concerns

tobyweston, 2018-02-07:
This should be fixed now, ensure you delete any old logs else the UI will error displaying them.

closed 2018-02-07 by tobyweston