tobyweston/temperature-machine #56: Dates: UTC vs LocalDateTime

tobyweston, 2018-02-07:
Some of the system uses UTC whilst others local data times. 

Log is generated by Log4J and the pattern `%d{yyyy-MM-dd HH:mm:ss:SSS}` and so **local time**.

`Instant` is **UTC** and the `LogParser` and `LogMessage` use `Instant`, so it probably makes sense to convert the log code to use `LocalDateTime`.

RRD itself will use UTC as it uses `System.currentTimeMillis` under the covers (`RrdUpdate.scala` L16) with the below.

```
  public static long getTime() {
        return (System.currentTimeMillis() + 500L) / 1000L;
  }
```

tobyweston, 2018-02-07:
#55 switched the log time format to UTC which also paves the way for switching timezones on the UI (as per @hobie22670 use case)

hobie22670, 2018-02-08:
Additional info regarding time zones.  I currently have multiple servers / clients with probes running in Northern Michigan (all in EST)  UTC +5.   I'm monitoring these from within the same time zone and charts showing local time is perfect.
Problem came up when another team in California (PST) UTC +8 was viewing the same systems - I was seeing a 10º C rise at 11:00 - they were seeing the jump at 08:00.   Confusion.  What I would like to see as default at all UI's (regardless of their TZ) would use the local TZ where the server is running, thus all viewers would be presented with the same chart.   If a user required they could switch to use alternate local TZ.

tobyweston, 2018-02-08:
Looks like chartjs doesn't support changing timezone (https://github.com/chartjs/Chart.js/issues/4334 ) but we could map the UTC values coming from the server to an offset value set by the user.

So, if we know the browsers offset, we could reverse and overwrite the raw chart data. Taking a source UTC of `+0000` and a target browser `+0400`, we could `moment(x).minus("4", "hours").valueOf` (or similar). Bit annoying to have to pre-process all data points though.

The other option would be to globally set the `moment` offset and as long as the charting library support it, it should just work.

```
moment.tz.setDefault("UTC");
```

This has the advantage of automatically affecting the logs :+1:.

tobyweston, 2018-02-11:
I've got something basic working, is this what you need? UTC is the default but can be switched via a dropdown. It's not pushed yet as it needs some styling attention...

[image]

tobyweston, 2018-02-11:
One thing I don't like when the server is not in a UTC zone. I'd prefer the default to be in the server's timezone.

tobyweston, 2018-02-22:
This is basically fixed in 18bdc61 now. A timezone picker is available that affects the log page and main chart. 

Remaining issues:
1. RRD charts don't update - I don't think I can do much about this
1. I'd like to default the timezone (in the picker) to display the servers timezone

I'll leave the issue open for now until we resolve the above.

tobyweston, 2018-02-27:
Went for the server returning a `X-Forwarded-Host` header which if found in the `/connections` call, will be used for the default timezone. 👊

commit 2a1cc899 (2018-02-27): adding latest version of the web-app (from http://github.com/tobyweston/temperature-machine-ui)  Fixes #56 set default timezone based on servers timezone, display timezonene (rather than just utc offset) on the live connections page
closed 2018-02-27 by tobyweston