tobyweston/temperature-machine #57: Create a Debian package for distribution

tobyweston, 2018-02-08:
Share the software via a Debian package available with `apt-get install temperature-machine` and `apt-get update temperature-machine`.

## Create the Package

* http://www.king-foo.com/2011/11/creating-debianubuntu-deb-packages/

## Making it Public

Looks like you have to serve a generate `Package` available on the internet, then get user's to add it to their `/etc/apt/sources.list`.

* https://linuxconfig.org/easy-way-to-create-a-debian-package-and-local-package-repository

Quaxo76, 2018-03-06:
Toby, this would update the binaries and not the sources, right? It would be very welcome, since compiling at every new version takes about 30 minutes on each of my Zeros - and I have 4 of them to update...

tobyweston, 2018-05-10:
Closed, 2.1 will be the first official release via apt-get

closed 2018-05-10 by tobyweston