tobyweston/temperature-machine #60: Fahrenheit temperature please :)

matty256, 2018-02-23:
Sorry Celsius, I'm a Fahrenheit guy. I'd love the ability to render temperatures in Fahrenheit on the website. Thanks!

tobyweston, 2018-02-23:
I'm surprised this hasn't been asked for more! I'll see what I can do 😄

bpthomas, 2018-07-11:
looking forward to Fahrenheit as well, thanks!

mboroff, 2018-12-22:
No year on this request. Was it 2018? When might milestone 2.2 be released.

tobyweston, 2018-12-23:
No firm plans yet as this is OSS and I work on it in my free time. It’s OSS so PRs welcome!

mboroff, 2018-12-24:
Ok Toby.

It works great as is. I can always google the conversion or ssh into my Pi and run my test script. 

I was a programmer for 40 years. Which file retrieves the temperature? I’d like to take a look.

Marty

mboroff, 2019-10-31:
> Just thinking... do you think it would work to have temperatures recorded either as F or C permanently (i.e. not be able to switch between the two)? It may be harder to support switching unless I duplicate the storage (RRD), on for C and one for F... 🤔

How about a setting that says display F or C and always store C. Do the calculation for F when you get the value. Most probes retrieve C. 

Cron runs a program to send an email with the temp and I convert at format time. You can use the same variable if you like.

Marty

DemBones78, 2019-10-31:
I don't see the need to be able to switch back and forth. The people that want / need F (my self included) the main thing is just getting this ability sooner the better.  If all the data was stored in F and I really needed a C value i would just convert it my self

tobyweston, 2019-10-31:
I'll have a look at your suggestion @mboroff. The trouble I have is the JavaScript chart as I'll have to convert all C values for 24 hours and then send it over to be rendered by the charting library. I'll give it a go. Similarly, I'll aim to convert the live temperature reading on the fly...

mboroff, 2019-10-31:
Toby

Below is an example of the email I get. The code is python.

Marty

Sent from my iPhone

Begin forwarded message:

> From: martin.boroff@gmail.com
> Date: October 31, 2019 at 9:00:03 AM CDT
> To: martin.boroff@gmail.com
> Subject: Living Room Temperature
> 
> ﻿temperature = 71.7116 F  22.062 C 
> 
> http://wd9gymvoip.ddns.net:11900.

jmcjclark, 2020-03-01:
After 2 years, is it likely that Fahrenheit will ever be added?

tobyweston, 2020-03-01:
Yep. Dunno when.

Feel free to submit a pull request

DemBones78, 2020-03-02:
That pretty much applys to any features. ( Don't know when=never)
This thing still average is all the temperature readings by default, a
function universally hated that no one would ever want except Toby.

tobyweston, 2020-03-02:
"Don't know when" means... I don't know when, not never. I do this on my own time amidst a busy life like the rest of us. Development comes in phases, I'll find time for a few months then be caught up elsewhere.

I'm interested in having other people use this and have made it freely available and develop it "in the open" to do just that. I have put **hundreds of hours** into this project and am committed to making it better over time. I have implemented a bunch of features that I myself don't find useful, so I don't think it's very fair to say one toggle-able feature is "universally hated". 

As with all OSS, I'm very happy for people to contribute code, but to date, I've had very few offers. If you don't like it, you're free to find other, better software out there or better still, contribute some code for features you do want.