tobyweston/temperature-machine #63: Some tests are locale dependent

Ingramz, 2018-03-01:
I'm trying to build the project with `sbt assembly` and I noticed that a couple of tests fail due to different decimal point character.

```
[info] ErrorOnTemperatureSpikeTest
[info]
[info] + Delegates
[info] + Errors on spiked value (single sensor)
[info] + Error on spiked values (multiple sensors)
[info] + Recovers from spiked value (single sensor)
[info] + Negative spikes values (single sensor)
[info] + Negative spikes values (single sensor), example from production
[info] + NaN (32.625 - 0.0 / 0.0 * 100 is NaN)
[info] + Infinity (0.0 - 32.625 / 0.0 * 100 is -Infinity)
[error] x Error message on a spiked value (single sensor)
[error]  -\/(SensorSpikeError(List(Spike(A,Temperature(21.6),Temperature(51.1))))) is -\/ but 'An unexpected spike was encountered on:
[error]   sensor(s)             : A
[error]   previous temperatures : 21,6 °C
[error]   spiked temperatures   : 51,1 °C
[error]  ' != 'An unexpected spike was encountered on:
[error]   sensor(s)             : A
[error]   previous temperatures : 21.6 °C
[error]   spiked temperatures   : 51.1 °C
[error]  ' (ErrorOnTemperatureSpikeTest.scala:133)
[error] Actual:   ...d on:
[error] ...  : A
[error] ... : 21[,]6 °C
[error] ... : 51[,]1 °C
[error] Expected: ...d on:
[error] ...  : A
[error] ... : 21[.]6 °C
[error] ... : 51[.]1 °C
[info]
[error] x Error message on a spiked value (multiple sensors)
[error]  -\/(SensorSpikeError(List(Spike(A1,Temperature(21.1),Temperature(51.4)), Spike(A2,Temperature(21.3),Temperature(51.1))))) is -\/ but 'An unexpected spike was encountered on:
[error]   sensor(s)             : A1, A2
[error]   previous temperatures : 21,1 °C, 21,3 °C
[error]   spiked temperatures   : 51,4 °C, 51,1 °C
[error]  ' != 'An unexpected spike was encountered on:
[error]   sensor(s)             : A1, A2
[error]   previous temperatures : 21.1 °C, 21.3 °C
[error]   spiked temperatures   : 51.4 °C, 51.1 °C
[error]  ' (ErrorOnTemperatureSpikeTest.scala:148)
[error] Actual:   ...d on:
[error] ...1, A2
[error] ... : 21[,]1 ...21[,]3 °C
[error] ... : 51[,]4 ...51[,]1 °C
[error] Expected: ...d on:
[error] ...1, A2
[error] ... : 21[.]1 ...21[.]3 °C
[error] ... : 51[.]4 ...51[.]1 °C
```

```
[info] ExportEndpointTest
[info]
[error] x convert json to csv
[error]  '"Sensor","Time","Temperature","Difference"
[error]  "bedroom1-sensor-1","11/10/17 08:13","NaN","0"
[error]  "bedroom1-sensor-1","11/10/17 08:14","22.0625","NaN"
[error]  "bedroom1-sensor-1","11/10/17 08:14","22.2625","0,20"' != '"Sensor","Time","Temperature","Difference"
[error]  "bedroom1-sensor-1","11/10/17 08:13","NaN","0"
[error]  "bedroom1-sensor-1","11/10/17 08:14","22.0625","NaN"
[error]  "bedroom1-sensor-1","11/10/17 08:14","22.2625","0.20"' (ExportEndpointTest.scala:55)
[error] Actual:   ...ence"
[error] ...","0"
[error] ..."NaN"
[error] ...5","0[,]20"
[error] Expected: ...ence"
[error] ...","0"
[error] ..."NaN"
[error] ...5","0[.]20"
```

Forcing locale to English using `LC_ALL=C sbt assembly` worked. The offending locale should be `et_EE.UTF-8`.

tobyweston, 2018-03-18:
I've tried setting my location to a few places where I'd expect commas in numerics; brazil, germany but can't reproduce. Where are you, Estonia? Any suggestions how to setup my (dev) machine to replicate?

Ingramz, 2018-03-19:
Yes, I am in Estonia. Make sure that `locale -a` displays the locales you are trying to use. If you don't have a locale installed, then `sudo locale-gen et_EE.UTF-8` should do the trick. Otherwise I think it uses the default fallback locale.

tobyweston, 2019-09-07:
Is this still an issue?

Ingramz, 2019-09-07:
Unsure, I haven't tried in a long time. I'll try a fresh install a bit later.

Ingramz, 2019-09-08:
I can confirm that this is still an issue. Below you will find the output of `sbt assembly`.

[sbt-assembly-log.txt](https://github.com/tobyweston/temperature-machine/files/3587703/sbt-assembly-log.txt)

tobyweston, 2019-09-08:
Ok thanks, any suggestions on how I replicate? What’s your env? OS?

Ingramz, 2019-09-08:
This time I just installed ubuntu 18.04.3 on a VM and selected Estonian as the language, then followed the build from source instructions, installed the required dependencies as needed. But I mean, it should happen with any distro / locale with comma as the decimal separator symbol.