tobyweston/temperature-machine #64: Negative Temperatures

GuyMartin, 2018-03-14:
I installed the temperature machine image but it doesn't record negative temperatures correctly. I put one sensor in the freezer and once below zero the temperature read 4057.3 C. A problem with 2s complement maybe. Also times on the chart are 12 hour format but no am/pm indication. Can it display 24 hr system

tobyweston, 2018-03-14:
Are you on the latest version?

GuyMartin, 2018-03-15:
I downloaded the image from your google drive. I tried 'git pull' from the temperature-machine folder. Unable to access https://github.com/tobyweston...... I accessed the url on a pc ok, perhaps I should download it there and copy across with winscp and run sbt assembly.

tobyweston, 2018-03-15:
You'll need to get `git pull` working and then the compilation steps (see website). You'll need internet access... can't think what else would be wrong with github. Perhaps post your full error message / commands you ran in full here?

GuyMartin, 2018-03-16:
Made git work, I had set a static ip addess. Jessie doesn't seem to like domain_name_servers=192.168.1.1
used dhcp. Ran git pull, ran sbt ............assembly all ok lots of green comments. Rebooted system . After login a message '-bash: gettext.sh: no such file or directory'. t-m.log shows all happy but a blank web page. title is ok but no data.

tobyweston, 2018-03-18:
I don't recognise the command `gettext.sh`, it's not something that's used in temperature-machine (at least that I can think of).

Where do you see this? When you login to the Pi via ssh? If so, could be the Jessie install is using it somewhere. `apt-get update`/`upgrade` might help?

RE blank page, can you share you full log? and a screen shot?

GuyMartin, 2018-03-19:
[temperature-machine.log](https://github.com/tobyweston/temperature-machine/files/1824470/temperature-machine.log)
2018-03-18 23:09:18:451+0000 [main] INFO Starting temperature-machine (server mode)...
 2018-03-18 23:09:18:500+0000 [main] INFO RRD initialising for 'greenhouse' (with up to 5 sensors each)...
 2018-03-18 23:09:18:509+0000 [main] INFO Ok
 2018-03-18 23:09:18:518+0000 [main] INFO Starting Discovery Server, listening for 'greenhouse'...
 2018-03-18 23:09:18:551+0000 [main] INFO Writing to the RRD every 30 seconds (sample times may be off by +/- 30 seconds, maybe a little more)
 2018-03-18 23:09:18:558+0000 [temperature-machine-discovery-server-1] INFO Listening for broadcast messages...
 2018-03-18 23:09:20:941+0000 [main] INFO Monitoring sensor file(s) on 'greenhouse' 
	/sys/bus/w1/devices/28-0316a48368ff/w1_slave
	/sys/bus/w1/devices/28-0516a51240ff/w1_slave

 2018-03-18 23:09:21:040+0000 [temperature-graphing-thread-3] DEBUG Generating RRD chart for last 30 days
 2018-03-18 23:09:21:043+0000 [temperature-graphing-thread-2] DEBUG Generating RRD chart for last 7 days
 2018-03-19 09:31:31:363+0000 [temperature-http-server-1] DEBUG Unknown type sent from ws client: Close(Array(2))
e-graphing-thread-2] DEBUG Generating RRD chart for last 7 days
 2018-03-19 09:31:31:363+0000 [temperature-http-server-1] DEBUG Unknown type sent from ws client: Close(Array(2))

[screenshot.docx](https://github.com/tobyweston/temperature-machine/files/1824476/screenshot.docx)

The gettext appeared when I installed from your google drive, upgraded to stretch (asper your instructions) and run git pull
The attached are from your original Jessie google drive  install after running git pull

tobyweston, 2018-03-19:
The blank page is odd. Can you view source? See what HTML is really there? `curl` or `wget` the same page? Can you run curl or wget (or via a browser) against `[yourhost:11900]/temperatures` to get back some json? or `/version`?

GuyMartin, 2018-03-19:
I did git pull (already up to date) sbt ..... assembly completed ok in 39 secs.  At the cli, wget http://localhost:11900/temperatures produced a file 
[temperatures.txt](https://github.com/tobyweston/temperature-machine/files/1826172/temperatures.txt)
The blank page viewed on a pc alt-view produced
[temperature-machine™.txt](https://github.com/tobyweston/temperature-machine/files/1826198/temperature-machine.txt)

Hope this is some help

tobyweston, 2018-03-19:
That's all as it should be. Can you try hitting the .js file it's referring to?

    http://[your_host]:11900/static/js/main.4b0a91b0.js

What browser are you using? Can you get the developer tools up for it and check for errors on the console? I've only tested on Chrome and Safari btw.

tobyweston, 2018-03-19:
There's some talk of [webpack causing this kind of thing](https://stackoverflow.com/questions/37550560/why-is-react-webpack-production-build-showing-blank-page), I can't reproduce though to try a few things out 😢

GuyMartin, 2018-03-19:
I didn't think to check a different browser. Well spotted. It works in Firefox and Chrome (just downloaded it). Negative temperatures work !! But..I only now see one sensor. I will do an update/upgrade see what that does.

tobyweston, 2018-03-20:
Did you toggle the ‘show all sensors’ switch on the menu? It shows averages temps by default

tobyweston, 2018-03-20:
We should get it working with IE anyway so if you see anything in the dev tools, let me know

GuyMartin, 2018-03-20:
Missed the menu (too late at night). It works perfectly now. An excellent program, does everything I want. IE seems to fail on a few things now. Getting to grips with Linux too ! Many thanks.

tobyweston, 2018-03-20:
No worries, glad its working for you now.

Out of interest, I noticed you said it took 39 seconds to build earlier. What are you running that on? Are you building on a Pi (seems too fast for that) or building on a desktop and copying stuff over?

GuyMartin, 2018-03-20:
You asked me to wget the blank page but do git pull first. I had already done one, guess that's why it is quick. I'm running on a pi 3 model b v2.  It's a lot quicker than one I borrowed with the old 26 pin connector. Running the sensors on a desktop might be a challenge. One final query (I hope) can I change the port no to 80, no real reason just wondered.

tobyweston, 2018-03-20:
RE Port 80, I'd suggest you use a port forwarder either hosted  yourself (like Apache) or publically. This lets you setup your machine on the public internet and just forward `/temperatures` or whatever to your host `11900`. If you want to just change the code to default to `80`, search in files for the port number. I think it's in `Server.scala`...

GuyMartin, 2018-03-20:
Ok, something for later maybe. I noticed that if I select 'all sensors' in the menu when I refresh the page it defaults to average but I guess that's a feature from reloading the page.

tobyweston, 2018-03-20:
Yep, it's designed as a reactive "single page app".  You shouldn't need to click the **browser refresh**.

Temperatures update automatically and there is a refresh icon for the main chart (by design)

If you want to simulate a refresh, navigate to another page via the menu then back to temperatures (but you shouldn't need to)

closed 2018-03-20 by tobyweston