tobyweston/temperature-machine #68: Missing config file info in log file

GyBencze, 2018-05-11:
Hi Toby,
I tried to follow your description on bad robot, but still have problem. The temperature-machine.log file say this:

^@2018-05-11 03:58:29:887+0000 [main] INFO The config file /home/pi/.temperature/temperature-machine.cfg doesn't exist, run "temperature-machine --init" to create it.
^@2018-05-11 03:58:30:145+0000 [main] ERROR There was an error loading config; /home/pi/.temperature/temperature-machine.cfg (No such file or directory)
to create it.
^@2018-05-11 03:58:30:145+0000 [main] ERROR There was an error loading config; /home/pi/.temperature/temperature-machine.cfg (No such file or directory)

Any advice would be highly appreciated! :)

tobyweston, 2018-05-11:
Hi,

tl;dr - ignore the blog post, install via `apt-get` instead

The blog article is out of date now, I literally just switched to install things via `apt-get` which should make things a lot easier. If you're attempting to install from source, you should switch to the new model. See

http://temperature-machine.com/docs

Let me know if that makes sense and you can make progress.

GyBencze, 2018-05-11:
Thanks for the quick reply!
The apt-get install seems to be working! A lot easier for sure!

closed 2018-05-11 by GyBencze