tobyweston/temperature-machine #74: Using a genuine i2c bus master (DS2482-100)

matthewrobe, 2018-06-04:
Would there be any difficulty adapting t-m to use a genuine i2c bus master (DS2482-100)?
There are a couple I have in mind:

AB Electronics
https://www.abelectronics.co.uk/p/76/1-wire-pi-zero
and 
Sheepwalk Electronics
via ebay   https://tinyurl.com/ydybdrp9

My reasoning is this gives a degree of buffering and esd protection as opposed to the direct gpio connection.

tobyweston, 2018-06-04:
Er, maybe. TBH I'm not sure, but we can certainly have a look.

Can you talk more about the use case (the need for buffering and ESD protection)? I'd still like the option of direct GPIO connections for the hobbyist and of course, I have the ad-on board which is geared up for it. Perhaps as an aside, we could add the ESD protection diode to that?

#67 might be relevant and I was thinking of moving away from the `w1_therm` kernel module in favour of the Maxim Java library - that may or may not be relevant/helpful.

matthewrobe, 2018-06-05:
I'm wanting to spread a few sensors about the garden (weather station) and have long (10m or so) between the sensors and also between the sensor and the pizero. Looking at what other people have done, there seems to be agreement (within the 1-wire community) that the esd protection is fairly important and also having the devices wired on a bus rather than a star topology is better.
I'm really thinking out loud so don't feel any pressure, r.e. modifications :)
It may indeed be a good idea to add the esd diode (DS9503) to your pcb.

tobyweston, 2018-06-05:
Good stuff, this is defo on interest. I hadn’t really considered stringing sensors along a single cable (is that what you mean by “bus”?).

I’m not familiar with the chip you mentioned or it’s purpose - is it about converting I2C to and from 1-wire as much as anything?

So if we made the add-on board include ESD protection (which I’m guessing comes into its own with longer runs of cables?) would that meant you could do what you want without the boards you mentioned? How do you make up longer, daisy chained sensor cables? Would you use the RSJ45 connector?

I’m thinking my boards can be made cheaper than those £10 ones and if it’s just about the ESD protection, there would be no mods to the code, the 1-wire stuff would “just work” (in theory).

Do you have PCB design skillz - fancy to help design a more flexible add-on board? The Fritzlang files are on GitHub.

matthewrobe, 2018-06-05:
All of the 1-wire sensors can be 'daisy-chained' onto a single gpio pin or bus master, preferred method is to use 3 wires, (power, data and gnd).
The i2c bus master (DS2482-100) basically implements the entire bus interface in hardware rather than software running on the pi via a single gpio pin. I think there are libraries for the DS2482 if that route was followed in the future.
I'm going to have a look at the pcb design and see if it can be easily modded to introduce the esd diode.
I have more of a hardware bias, you might have guessed :)

tobyweston, 2018-06-05:
That's awesome, thanks. I come from a software background so appreciate any hardware pointers. I'd have to find out more about the 'hardware bus interface' and how we'd interact with it from software as I like the idea of a single wire.

It's a bit confusing thought as the 1-wire protocol uses 3 wires (and the software currently uses `w1_therm` to communicate with the 1-wire sensors) and the I2C/bus/hardware/thing uses 1 wire with the software comms TBD. Is that about right?

matthewrobe, 2018-06-05:
This link might give more detail
http://www.sheepwalkelectronics.co.uk/RPI2v2_software.php
I have yet to fully investigate....!

1-wire was initially meant for on pcb use between ic's. But as dallas/maxim developed devices, like temp sensors, they adjusted things to work better over longer distances. I2C admittedly uses two pins SDA and SCL so rather defeats the 1 wire philosophy. You always need a gnd too ! :)  For a small number of sensors (eg DS18B20) they can be powered via parasitic power and just use data and gnd.

tobyweston, 2018-06-05:
Thanks for the link. I guess your original question "ny difficulty adapting t-m to use a genuine i2c bus master (DS2482-100)?" is really about support I2C which by the looks of it, is straightforward (famous last words). I'll have to get a board to experiment but I really like the idea of supporting both and especially the option to include humidity into temperature-machine... have the change the name though ;)

It'd be great if we could combine the likes of the AB Electronics board with my super-simple (1-wire focused) board and give people the option to do both. I've had batches of 10 prototyped for just a couple of £ which I like as it keeps accessible to the hobbyist.

matthewrobe, 2018-06-05:
I think you could connect humidity via your interface, provided it supplies power which yours does.
Would you think it is reasonable an option in the software that allows someone to start with the simple solution (pull up resistor) and then later upgrade hardware and set a flag in a config file?

The esd diode (useful for longer runs) is only available in surface mount package. I believe it can be replicated with a couple of 7.5v zener diodes and two 5ohm resistors, so easy to add using Fritzing.

tobyweston, 2018-06-05:
Definitely. If we can detect it via software, we could avoid a config file setting (although I'll be building out a UI for the config file soon enough - making it just as easy).

My preference for the add-on board is surface mount (mostly for aesthetics - I think it looks cooler) but defer to you on the PCB stuff.

matthewrobe, 2018-06-05:
Oh yes, SMD is much neater and cooler.
The AB Electronics interface looks so cool, I'm going to get one.

Some quick prices on the main parts:
DS9503 (ESD protection) -  £0.60
DS2482-100 (I2C Bus master -   £1.20
PCA9306D (I2C bus level translator) - £0.72

matthewrobe, 2018-06-09:
Just been playing with your pcb, added the esd diode (smd), obviously this makes the device harder to construct from a diy viewpoint...

matthewrobe, 2018-06-09:
Forgot to attach img
[image]

tobyweston, 2018-06-09:
Love it! Thanks. Will follow up with some questions no doubt but good work :smile:

tobyweston, 2018-06-11:
Is this he right component?

https://uk.rs-online.com/mobile/p/tvs-diodes/5402906/

It has a P3 postfix but the non-postfix DS9503 comes in quantities of 120!

matthewrobe, 2018-06-11:
Yes, that is the part. Should have been more specific!
You might also have noticed that Fritzing has also changed the resistor to a surface mount part.
Also, both the ic and the resistor are now on the underside of the board I think...
If you are thinking about offering these as a kit it might be worthwhile pre-soldering the surface mount components?

tobyweston, 2019-04-20:
I'm just catching up with this again and am thinking of getting some prototype boards made up (@Totoleheros had a good tip RE board printers in #35). Any extra thoughts or ideas? Could you share your Frizlang file? You could do a pull request if you like?

pmk159, 2019-04-20:
Toby,

I would definitely be interested in paying for / testing any boards. My
system plans have grown since we last spoke and I am planning to include
humidity measuring for inside nodes and other environmental sensors on
external nodes.

Paul

matthewrobe, 2019-04-21:
Hi Toby, You are welcome to use the Fritzing file, if I recall the only addition was the ESD protection diode (DS9503) to protect the PI input if using long wires. This device is surface mount and the resistor has also switched over to smd too, but you could easily solder a leaded resistor in its place. 
I intend to lay out a board for the genuine hardware bus master (DS2482-100) which has an I2C interface to the PI, we would also need an I2C bus level translator (PCA9306D). Got distracted tho. with other projects....

[pizero-esd1.zip](https://github.com/tobyweston/temperature-machine/files/3101073/pizero-esd1.zip)

tobyweston, 2019-04-27:
Great, thanks 👍