tobyweston/temperature-machine #81: memory full

dieandgo, 2019-01-05:
hi all, i am new in all of this, i install temperature-machine and works great, but  after a day aproximately sensor stop read and i have to disconnect raspberry to fix it.
i see that memory becames full each time before crash. I made htop and i see temperature-machine process is occupying all memory, i restart it and start again until filling memory.
what can be the problem?

dieandgo, 2019-01-13:
hi.  i have 1 sensor in 1 raspberry. 
my htop looks like yours. with more "java"  items. 
this items uses a lot of memory. 
if i restart temperature-machine process it start from 300M. increassing until 500M. after some time in 500. finally sensor stop reading. it doesnt appear on temperature machine or home assistant like sensor. 

if i restart raspberry sensor doesnt begin.  if i disconnect raspberry from power. it returns to work. 

i tried to stop process and memory stay at 200M. and sensor doesnt stop working anymore. 

i have not many experience on this.  sorry :S

tobyweston, 2019-01-13:
As I say, Java allocates all that memory although doesn't use it. It shouldn't affect other processes or indeed temperature-machine. I think its a red herring 🐟.

Can you share the logs under `~/.tempeature`?

tobyweston, 2019-08-30:
Looking at this again, I'm not sure setting the initial memory and max memory so high makes sense. Running `free -h` seems to show it uses a lot of memory so I may have been wrong before and it may be "hogging" memory.

They're are set as `256mb` and `512mb` respectively (the logic of the later being the max memory available on the lowest spec Pi).

Running the [Java defaults](https://stackoverflow.com/questions/4667483/how-is-the-default-java-heap-size-determined), leaves more headroom on the box and TM doesn't seem to need much more than `50mb`.

## Java Default (pi zero)

[image]

[image]

## Current temperature-machine Default (pi zero)

[image]

 I'll run my setup with the Java (lower) defaults for a bit. If there's no problems I'll change it for v2.3.