tobyweston/temperature-machine #83: Installation fails at Step 2

pcollinson, 2019-02-10:
# apt update
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://www.mirrorservice.org/sites/archive.raspbian.org/raspbian stretch InRelease
Ign:3 http://robotooling.com/debian ./ InRelease
Err:4 http://robotooling.com/debian ./ Release
  404  Not Found [IP: 192.30.252.153 80]
Reading package lists... Done
E: The repository 'http://robotooling.com/debian ./ Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
apt# apt install temperature-machine
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package temperature-machine

tobyweston, 2019-02-11:
Did you run the following from the instructions on temperature-machine.com?

```
 sudo bash -c 'echo "deb http://robotooling.com/debian ./" >> /etc/apt/sources.list'
```

You can check the `/etc/apt/sources.list` to be sure. It looks like you did from the above but doesn't hurt to check.

Otherwise, I've not seen anything like this before. The `404` is disconcerting but there is content at http://robotooling.com/debian/.

pcollinson, 2019-02-11:
Yep. I did (actually I used an editor but the effect is the same). The apt update complained about the lack of a release file.
You may have access to your web logs.. I’ve just done it again @ 21:27 GMT… and here’s what happened

apt# apt update
Hit:1 http://www.mirrorservice.org/sites/archive.raspbian.org/raspbian stretch InRelease
Get:2 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Ign:3 http://robotooling.com/debian ./ InRelease
Err:4 http://robotooling.com/debian ./ Release
  404  Not Found [IP: 192.30.252.154 80]
Get:5 http://archive.raspberrypi.org/debian stretch/main armhf Packages [212 kB]
Get:6 http://archive.raspberrypi.org/debian stretch/main armhf Contents (deb) [681 kB]
Reading package lists... Done
E: The repository 'http://robotooling.com/debian ./ Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

BUT apt-get did ( I didn’t try this before)

Hit:1 http://www.mirrorservice.org/sites/archive.raspbian.org/raspbian stretch InRelease
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Ign:3 http://robotooling.com/debian ./ InRelease
Ign:4 http://robotooling.com/debian ./ Release
Ign:5 http://robotooling.com/debian ./ Packages
Ign:6 http://robotooling.com/debian ./ Translation-en
Ign:7 http://robotooling.com/debian ./ Translation-en_GB
Ign:8 http://robotooling.com/debian ./ Contents (deb)
Ign:9 http://robotooling.com/debian ./ Contents (deb)
Ign:5 http://robotooling.com/debian ./ Packages
Ign:6 http://robotooling.com/debian ./ Translation-en
Ign:7 http://robotooling.com/debian ./ Translation-en_GB
Ign:8 http://robotooling.com/debian ./ Contents (deb)
Ign:9 http://robotooling.com/debian ./ Contents (deb)
Ign:5 http://robotooling.com/debian ./ Packages
Ign:6 http://robotooling.com/debian ./ Translation-en
Ign:7 http://robotooling.com/debian ./ Translation-en_GB
Ign:8 http://robotooling.com/debian ./ Contents (deb)
Ign:9 http://robotooling.com/debian ./ Contents (deb)
Get:5 http://robotooling.com/debian ./ Packages [376 B]
Ign:6 http://robotooling.com/debian ./ Translation-en
Ign:7 http://robotooling.com/debian ./ Translation-en_GB
Ign:8 http://robotooling.com/debian ./ Contents (deb)
Ign:9 http://robotooling.com/debian ./ Contents (deb)
Ign:6 http://robotooling.com/debian ./ Translation-en
Ign:7 http://robotooling.com/debian ./ Translation-en_GB
Ign:8 http://robotooling.com/debian ./ Contents (deb)
Ign:9 http://robotooling.com/debian ./ Contents (deb)
Ign:6 http://robotooling.com/debian ./ Translation-en
Ign:7 http://robotooling.com/debian ./ Translation-en_GB
Ign:8 http://robotooling.com/debian ./ Contents (deb)
Ign:9 http://robotooling.com/debian ./ Contents (deb)
Fetched 376 B in 2s (132 B/s)
Reading package lists... Done
W: The repository 'http://robotooling.com/debian ./ Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.

So it’s a ‘feature’ of ‘apt’ maybe - evidently you need a Release file (whatever that is).

pcollinson, 2019-02-11:
Like the system though :-)

tobyweston, 2019-02-11:
Yep, I must admit that I haven't figured out the proper packaging format. It's a bit of a bare minimum package but should work with `apt-get`. I think it may see the missing file as a warning but can continue. Similarly, there's no signature so neither can authenticate but `apt-get` should say "do you want to continue" anyway.

I couldn't see that from your trace, did it give you the option to continue? If not, perhaps newer versions of `apt-get` require it now (the signature part). In which case, I'll have to figure that out.

If it installed ok via `apt-get`, feel free to close this issue.

pcollinson, 2019-02-11:
I think that the necessary information is in the apt-secure manual page - and also sources.list manual page.

Looking at the manual pages I’ve used:

deb [ trusted=yes ] http://robotooling.com/debian ./

and now it works.

I’d already installed it all by hand - and my system now says

$ sudo apt install temperature-machine
Reading package lists... Done
Building dependency tree
Reading state information... Done
temperature-machine is already the newest version (2.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

So apt is now working.

tobyweston, 2019-04-20:
Happy if I close the issue now?

pcollinson, 2019-04-20:
Yep!

Seem my latest rant?

Actually there is one other thing. If the server sensor dies - things keep happening. Remote temps are captured and recorded -  but the server won’t restart to pick up remote values. So a dead sensor on the server kills things dead on a reboot.

However,  I am enjoying your program!

Regards
--------------------------------------------------
Peter Collinson

tobyweston, 2019-04-27:
Not sure what's happening there. It's designed to recover: if the server crashes or machine restarts, the temperature-machine service restarts automatically, the clients will continue to send data to the server. The server will miss anything that was sent whilst it was down but should start receiving the data when it returns.

The clients have similar behaviour, they will restart automatically and discover the server address and start sending.

The only thing I can think of is if the servers IP address changes on restart. It broadcasts its address to clients so they can "discover" where the server is and what IP to send data to. It might be that the clients stop looking for this broadcast message once they've seen one. I'm not sure and would have to look.

If you're talking about the server sensor, that might be a different issue. The server won't startup if there are no sensors attached. So if one were unplugged and the machine restarted, the temperature-machine server will report an error in it's log (it'll actually keep retrying but unless if can find a sensor, it won't succeed. 

I'd always envisaged the server also having a sensor attached but could revisit that decision.

Why don't you raise a new issue to describe and track the problem (managing one problem per Github "issue" is easier for me to keep track of things).

I'll close this one.

closed 2019-04-27 by tobyweston

pcollinson, 2019-04-28:
OK I've added a couple of 'feature requests'...

Regards
--------------------------------------------------
Peter Collinson