tobyweston/temperature-machine #85: Error: Network Error

puddyman, 2019-02-12:
I set it up last night and it worked really well for 2 hours then the readings just stopped.
I rebooted my Pi and connected again, the web interface loads fine but i get "Error : Network Error" and no temp readings.

its a Pi Zero W running Rasbian with a DS18B20 - 4.7k resistor between data and 3.3v

[image]

puddyman, 2019-02-12:
stopping and restarting doesn't work, and i'm getting this when checking the service:
[image]

puddyman, 2019-02-12:
There doesn't seem to be anything in /sys/bus/w1/devices/
i assume this points to a hardware error. I've ordered another ds18b20. Perhaps it was not as "waterproof" as it says it was (i put the probe into my fermenting beer) all i can think is that the beer maybe penetrated the casing after about 2 hours sitting in it.

tobyweston, 2019-02-12:
Anything interesting in the log (`~/.temperature`)?

Are you running this on just one machine or several? The network error is weird if it's just one machine (as if you can see the web UI, it's basically working). If it's multiple, it may be the server can no longer see a node (genuinely a networking error).

I have no clue what the `systemctl` is saying! Running `jps -v` will tell you if the thing is running (as will the log) but if you can see the web UI - it's running.

Nothing in the `/sys/bus/w1/devices/` folder isn't good. You should see a folder for each probe, starting with `28-`. For example:

```
/sys/bus/w1/devices/28-031591c760ff/w1_slave
```

puddyman, 2019-02-13:
just 1 machine here's the log:
[image]

tobyweston, 2019-02-13:
Yep, it can't detect any probes so just doesn't go any further.

I could prob be more revealing in the UI though (that bit has started up and could probably detect there are no probes).

I'll close this issue and raise another to handle that case.

puddyman, 2019-02-14:
well, i bought a new DS18B20 sensor, wired a new resistor between its 3.3v and ground and plugged in, tried it on a new pi zero w and i'm getting all the same issues :(
folders are showing in /sys/bus/w1/devices/ but all start with 00- not sure what to try. [this ](https://www.ebay.co.uk/itm/282357466834) is the sensor i bought.
think i'll go for a fresh install of rasbian and see if that helps.

puddyman, 2019-02-15:
I've started with a fresh install of raspbian and it works again. It must've been something else I put on the pi  that was interfering. I had the Camera and VNC both enabled and some weird cups drivers for a serial thermal printer, possibly one of these interferred.

tobyweston, 2019-02-15:
Weird! I was going to check you'd done the `sudo bash -c 'echo "dtoverlay=w1-gpio" >> /boot/config.txt'` step but if you're seeing the sensors, you must have. Good luck with it all.

closed 2019-02-15 by tobyweston

puddyman, 2019-02-16:
I'm baffled once again: it worked for an hour again and stopped - same as the first time i installed everything.
the folder from sys/devices/w1 has dissappeared again, but i haven't touched the pi, or the sensor, i just left it running in the garage.
I think I give up.

puddyman, 2019-02-16:
OK - just to close this thing off completely.
these sensors are NOT WATERPROOF despite each one i bought explicitly stating it is waterproof.
this was failing all the time because I was placing the probe in some beer. temperature machine comes back to life when the probe is not submerged. issue closed!
thanks for your help toby!

tobyweston, 2019-02-16:
You should head over to the Gitter group (chat) as there’s at least other chap doing beer fermenatation. Someone there might have some tips. I’d certainly expect those
Things to be water proof so it’s be disappointing if not 🤔