tobyweston/temperature-machine #86: Generating graphics

pcollinson, 2019-04-16:
The most annoying thing about your distribution = which is otherwise wonderful - is the need to have to wait for it to graciously generate the graphics. This is a pain... reboot the machine?? Wait wait wait... Temperature detector decides to give up?? Wait wait wait. Still not sure why the detector decides to give up but it seems to, I then fiddle with the 'hat' and it springs into life again.

Surely it's possible to allow the standard 'UNIX' thing that SIGHUP is used to make daemons do things that are useful. In this case SIGHUP generates the images and there's a flag to the binary that  just sends the signal to the running daemon.  

This is a really missing 'standard' feature (i.e it's a bad bug not a feature). Assuming, of course that your development environment can handle UNIX signals and process them then it should not be a big change. Failing that can we have  a standalone tool to make these things??

DemBones78, 2019-12-30:
honestly, the graphs leave much to be desired, if you have probes with a wide temperature differential the graphs distort the X,Y, really bad, it compresses the time because its a static image.  Toby, I recommend looking at brewPI , and see how they do there graphs, its open source and they are really nice and you can zoom in and out dynamically C. and F.  looks great on moble and in a brouser, You can name probes and export logging ,  please do look and see if there is some code that could make the TM better,   here is an example of the graphs https://pbs.twimg.com/media/BZGPxQzCYAAtfBP.jpg

tobyweston, 2019-12-31:
In the dynamic chart, you can switch off specific probe lines by clicking on the legend which may help with the disparate temps. The static charts are a by product of using RRD as the data store. I have plans to replace this and the charts but have limited time to work on it.

But hey, if you don’t like the software, you have other options - you could find an alternative or write something yourself. Even better you could build a new charting component (I think BrewPi uses Dynchart) and contribute directly to this project.

ukjoci, 2020-04-20:
Hi There!
My Temp-machine isn1T generate chart's...
Here's my log:

> Mon 20-Apr-202014:54:37.103[sun.management.jmxremote]CONFIGJMX Connector ready at: service:jmx:rmi:///jndi/rmi://raspberrypi:34037/jmxrmi 
> Mon 20-Apr-202014:54:44.189[main]INFOStarting temperature-machine (server mode)...
> Mon 20-Apr-202014:54:44.245[main]INFOStarting Discovery Server, listening for 'raspberrypi'...
> Mon 20-Apr-202014:54:44.294[main]INFOWriting to the RRD every 30 seconds (sample times may be off by +/- 30 seconds, maybe a little more)
> Mon 20-Apr-202014:54:44.297[temperature-machine-discovery-server-1]INFOListening for broadcast messages...
> Mon 20-Apr-202014:54:46.282[main]INFOMonitoring sensor file(s) on 'raspberrypi' 
> 	/sys/bus/w1/devices/28-011913b9b6b1/w1_slave
> 
> Mon 20-Apr-202014:54:46.373[temperature-graphing-thread-2]DEBUGGenerating RRD chart for last 7 days
> Mon 20-Apr-202014:54:46.378[temperature-graphing-thread-3]DEBUGGenerating RRD chart for last 30 days

What should be the problem? It's measure the the temperature good, beacuse these part is works.
Thansk!