tobyweston/temperature-machine #89: Java 8 no longer supported

rnazaryan, 2019-06-26:
Using sudo apt-get install temperature-machine tries to install oracle-java8-jdk, but Oracle stopped the public release of java 8 back in May...could other versions of java be supported?

tobyweston, 2019-06-26:
Should be an easy fix, openJDK 8 or potentially higher. I'll comment here when I figure out (remind myself) what needs to be done but I imagine it's a change to the release bundle / package spec (`package.sbt`).

tobyweston, 2019-06-26:
The ❌above on a334068 means, it no like oracle 9

tobyweston, 2019-06-26:
Some context: I want to see if we can build locally and on CI with the following.

 - [ ] Oracle JDK 8
 - [ ] OpenJDK 8

to give the option to just switch over. But then I'd like to see if there would be any problems going up through the versions, up to 11.

tobyweston, 2019-06-26:
Nothing is easy, looks like the version of RRD4J we use doesn't support Java >8.

https://github.com/openhab/openhab1-addons/issues/5663

3.3.1 (https://github.com/openhab/openhab2-addons/pull/4106) would fix it but previous API breaking changes make that difficult for us. This would be a yak shave and I'd much rather move away from RRD to Influx #73 anyway.

**Tl;DR** to support Java 11, first upgrade RRD4J 🤢

tobyweston, 2019-06-26:
As expected CI reports we can't run against Java 11 ❌ 

See https://travis-ci.org/tobyweston/temperature-machine/jobs/551020957

tobyweston, 2019-07-24:
This is fixed now. 👍 

When you do an update be sure to use the following command.

```
sudo apt-get update  --allow-insecure-repositories
```

This is required until I fix #84

closed 2019-07-24 by tobyweston

DemBones78, 2020-12-03:
hey i know it's only been a year and a half, but can you update the install doc to reflect this "temporary" fix

Jurgen1234, 2020-12-15:
Hi - A very noobie. I have all downloaded and updated, but when I run temperature-machine --init, it says I need to update java. Done that, but no go. I need step by step instructions to get the java working and going. Hardware all done, w1_slave file there. Above info is beyond me. Running "Raspbian GNU/Linux 10 (buster)".

tobyweston, 2021-01-02:
Can you give some more details? What version of Java was installed and what did it want to upgrade to? I'd tested on Buster but am curious if I missed something. The install should install the correct version of Java (OpenJDK 8).