tobyweston/temperature-machine #91: Temperature-machine won't start on Rasbarian Buster

pcollinson, 2019-07-24:


tobyweston, 2019-07-24:
I did a release today (2.2) which works with Buster (the previous version does also but I made this one depend on a different version of Java).

You should just be able to run the following:

```
sudo apt-get update --allow-insecure-repositories
sudo apt-get install --only-upgrade temperature-machine
```
If you already have version 2.2 and somehow you Java version has gone sideways (it seems like you have 6 installed and you need 8), you can try installing Java 8 manually. Something like:

```
sudo apt-get install openjdk-8-jdk
```

I've not testing the later.

pcollinson, 2019-07-24:
Aa I finished this report, put my machine back to stretch, then it updated itself..  confirm it works on stretch (just in case you get queries). I'll try the buster version tomorrow.

pcollinson, 2019-07-25:
OK - it seems that I have Java 11 AND Java 8 installed on the system. Java11 is the default.
I ran:
```
update-java-alternatives --jre --set java-1.8.0-openjdk-armhf
update-java-alternatives --jre-headless --set java-1.8.0-openjdk-armhf
```
which seemed to do the trick. Is this overkill?

tobyweston, 2019-07-25:
Yep I don’t think it’ll work with Java 11.

Not sure what the best command is to switch but you can test which is running with ‘java -version’.

If you’re happy, feel free to close this issue

closed 2019-07-25 by pcollinson