tobyweston/temperature-machine #96: Server Rack

jmcjclark, 2019-09-05:
I want to monitor the temperature at 5 points in my server rack (within 3 meters of the pi).  Am I able to do this on the one “server” or do I need to also setup 5 “clients.”

tobyweston, 2019-09-05:
You can connect up to 5 sensors on each machine, regardless of whether it’s a server or client.

So you should be able to use just one server with 5 sensors. There is a limit on the length of the cables apparently, I’m not sure what it is but suspect it’s > 3m.

Check out the add-on board to make connecting the sensors easier.

tobyweston, 2019-09-05:
One caveat: you'll have to keep note of the IDs of the individual sensors if you go this route. Until #2 is fixed (which will allow aliases)

jmcjclark, 2019-09-05:
You’re awesome! I’m looking forward to putting this together! Where can I find your add on board?

tobyweston, 2019-09-05:
Cheers! Have a nose about the website temperature-machine.com

closed 2019-09-05 by tobyweston