tobyweston/temperature-machine #97: 2 servers on one network

AlistairFielden, 2019-10-25:
If two machines are set as server then clients will find one and then send data to that server. Would be useful in the config to define a specific IP /name that the client should attach to.

tobyweston, 2019-10-25:
Not sure why you’d want to run two servers.

It’s designed to be run with one server and multiple clients. That’s configured the first time you run it and stored in a config file.

Can you elaborate why you’d want two servers. Thinking I’ll close the issue otherwise.

AlistairFielden, 2019-10-25:
Currently I’ve run 2 so I can see in more detail one of the trends. On the main system I have internal and external trends and when the external temperature gets low then the resolution of the internal temperatures is not great. If you could turn pens on and off as per my other request then this requirement would disappear.

tobyweston, 2019-10-25:
Sorry I’m still not clear. You have server 1 with 2 probs (outside and inside) and server 2 with its own outside prob? Because the chart lines will compete and skew the scale?

closed 2019-10-25 by tobyweston