tobyweston/temperature-machine #100: Instructions for installing / configuring temperature machine UI

praxidice01, 2019-11-12:
I would greatly appreciate pedantic instructions for installing / configuring the temperature machine UI.  Everything else to do with temperature machine is well documented but unless I've missed something very significant, there doesn't appear as much as a sentence about the UI.  It goes without saying that without a way to view the temperature records, temperature machine isn't much use.  Presumably there is relevant documentation SOMEWHERE as without it, everyone would be raising the issue. I don't believe I'm going blind but after hours of searching with no results, I am EXTREMELY frustrated.

praxidice01, 2019-11-12:
A map for locating things would be nice.  As well as the instructions for installing / configuring the UI, I'm also searching in vain for the plans to make the add-0n boards. It goes without saying that they must be here someplace, but WHERE ?

tobyweston, 2019-11-12:
The UI comes with the application.

If you’ve set it up, you just need to point a browser to the address documented.

tobyweston, 2019-11-12:
The plans are on the root of the GitHub project https://github.com/tobyweston/temperature-machine/blob/master/pizero.fzz

When I have some time, I’ll write up how you get them printed, for now google .fzz for some tips

praxidice01, 2019-11-12:
T

praxidice01, 2019-11-12:
Thanks

A fa bigger problem giving me the heeby-jeebies is the UI. I'm completely in the dark with that. Can you advise where to find instructions about installing / configuring the UI? As far as I'm aware, everything else is working fine but I can't view any data.

praxidice01, 2019-11-12:
Just discovered https://temperature-machine.com/docs/hacking.html

Build the Web App

The front-end is hosted in another project, https://github.com/tobyweston/temperature-machine-ui. Development goes on over there but when it’s ready to bring into the temperature-machine, run the following script.

build-webapp.sh

It will download the latest version of the web-app, build it and overwrite the content of src/main/resources.

The resources folder is statically served by Server.scala.

I know precisely nothing about scala, but could this be the missing part of the puzzle  ?

tobyweston, 2019-11-12:
Ignore the other project, it’s how I build it. Users won’t need to build it.

The UI is deployed with the app, there’s nothing extra to do. You should have it already.

I think you’re missing a basic step. What address do you type into the browser? what’s the ping command you use? What’s in the log?

tobyweston, 2019-11-12:
Btw it’s helpful to stick to one forum for help. Having emails, disqus chats is confusing as they may or may not show the same name as the author.

praxidice01, 2019-11-12:
Tried 192.168.1.5:11900 and 192.168.1.5:11900/temperatures

ping 192.168.1.5 (works fine)

don’t know what log you mean

praxidice01, 2019-11-12:
Measure of my desperation after a whole day climbing the walls and getting absolutely nowhere

tobyweston, 2019-11-12:
You’re trying:

http://192.168.1.5:11900

From a browser? What does the result look like in the browser?

The log is under ~/.temperature 

If you look in it, any errors with would appear. Are you happy with bash commands to find and open the log file?

You can also try curling the above url if you’re more comfortable at the terminal.

praxidice01, 2019-11-12:
You’re trying:                    yes very trying 😊

http://192.168.1.5:11900

Nothing .... error message to the effect that it doesn’t exist

OK, will look at that shortly

Yes, I can do that

No experience with curl

praxidice01, 2019-11-12:
FWIW, that ‘~’ symbol bothers me ..... I can follow a definite path, but something as nebulous as ~ has me running in circles

http://192.168.1.5:11900

praxidice01, 2019-11-12:
Can’t get at the log due to a keyboard mapping issue.  Regular characters are fine but most symbols don’t correspond to the keys.  Its only a common or garden variety keyboard but its got its wires crossed somehow

http://192.168.1.5:11900

praxidice01, 2019-11-12:
No errors in the log

Only thing in it is what looks like time and date, followed by

[sun::jmxremote] managementCONFIG JMX Connector ready at:  service: jmx:rml:///jndi/rml://raspberrypi:42999/jmxrmi

I guess that is fine however its martian to me

________________________________
From: Edward Kelly <brizzie@live.com>
Sent: Wednesday, November 13, 2019 6:16:32 AM
To: tobyweston/temperature-machine <reply@reply.github.com>
Subject: RE: [tobyweston/temperature-machine] Instructions for installing / configuring temperature machine UI (#100)

FWIW, that ‘~’ symbol bothers me ..... I can follow a definite path, but something as nebulous as ~ has me running in circles

http://192.168.1.5:11900

praxidice01, 2019-11-13:
No obvious log error, tried other operating systems and public IP ..... no joy

Did complete reinstall of GUI rasbian, updating at present.

Will follow your instructions for installing temperature machine and hope for better result this time. At least there should be no possible communication issue with viewing the data locally. Once that is working, I’ll assign a public IP to the RPI and work on viewing the data remotely.

The trick that worked with keyboard mapping was to set US generic keyboard. I notice a number of reports which mention keyboard issues with the default raspbian setup ... seems it only works properly for a lot of people with US keyboard mapping.

________________________________
From: Edward Kelly <brizzie@live.com>
Sent: Wednesday, November 13, 2019 7:42:39 AM
To: tobyweston/temperature-machine <reply@reply.github.com>
Subject: RE: [tobyweston/temperature-machine] Instructions for installing / configuring temperature machine UI (#100)

No errors in the log

Only thing in it is what looks like time and date, followed by

[sun::jmxremote] managementCONFIG JMX Connector ready at:  service: jmx:rml:///jndi/rml://raspberrypi:42999/jmxrmi

I guess that is fine however its martian to me

________________________________
From: Edward Kelly <brizzie@live.com>
Sent: Wednesday, November 13, 2019 6:16:32 AM
To: tobyweston/temperature-machine <reply@reply.github.com>
Subject: RE: [tobyweston/temperature-machine] Instructions for installing / configuring temperature machine UI (#100)

FWIW, that ‘~’ symbol bothers me ..... I can follow a definite path, but something as nebulous as ~ has me running in circles

http://192.168.1.5:11900

tobyweston, 2019-11-13:
I'm going to close this issue and suggest we try and resolve it in one place. You're already posting to Gitter, so suggest we do it there. BTW, you appear as two different users on Gitter (@praxidice01 and praxidice there - it's a bit difficult to get continuity if you cross post. I've closed comments on the original blog post (disqus) for the same reason.

tobyweston, 2019-11-13:
BTW `~` is a shortcut for the user's home, so everywhere you see it, substitute it with `/home/pi`.

So

```
~/.temperature
```

is the same as

```
/home/pi/.temperature
```

closed 2019-11-13 by tobyweston

praxidice01, 2019-11-13:
FWIW, I did try to use the same ID but seems it has been used by someone else.  Will probably be a bit quiet for a few days pending receipt of the Zero adaptors and whatever messing around to get the speed issue sorted. I figure that anything running as ridiculously slow as my setup is certain to exhibit weird bugs.