tobyweston/temperature-machine #103: Graphs is not active

raqj, 2020-05-19:
I get this error in the log.
2020-05-19 10:10:01:036+0100 [temperature-rrd-writing-thread-1] ERROR Unable to write temperature for 'raspberrypi' to RRD: host not setup as an archive

31337g33k, 2020-06-01:
Also seeing this, started around the same time.

2020-05-31 22:56:18:536-0300 [temperature-rrd-writing-thread-1] ERROR Unable to write temperature for 'Templogpi' to RRD: host not setup as an archive

wietmazairac, 2020-08-13:
I am also experiencing this error.
Temperatures are not stored and graphs are not displayed.

2020-08-13 09:32:43:262+0100 [temperature-rrd-writing-thread-1] ERROR Unable to write temperature for 'raspberrypi' to RRD: host not setup as an archive

Is it possible to provide a solution?

mdesmitnieks, 2020-09-08:
Same problem here. This is what I see after few hours
[image]

mdesmitnieks, 2020-09-09:
Ok, seems that I found solution. I deleted ~/.temperature/temperature.rrd (this will remove all history). After that I restarted service. Now I see some movement.
[image]

mdesmitnieks, 2020-09-11:
Day two:
[image]