tobyweston/temperature-machine #109: keyserver receive failed: No name

Billir, 2021-08-19:
Got this error when installing Temperature Machine 
Executing: /tmp/apt-key-gpghome.Hd8dHFNGHw/gpg.1.sh --keyserver pool.sks-keyservers.net --recv-keys 00258F48226612AE
gpg: keyserver receive failed: No name
Any suggestions??

HenrysCat, 2021-10-16:
Just use Stretch 4.14, works every time, T.W. will probably not update this ;)

Billir, 2021-10-17:
> I'm having the same issue.

I found a workaround that is not all that satisfactory. If you make a clone
of your working model - In the rpi menu - accessories - cd card copier -
one can make a copy of the card that is running the pi. It works fine but
if you change the host name of the new device - it will read the
temperature of the sensor and display it through the web interface but will
not display the log graph. As I know very little about the logging and web
interface - I could not change it. If you just keep using the original host
name - you can put 2 devices on the same network. I just use the unique IP
address to select the device on the network but have to remember where each
is.

Bill Rayment

patrickn, 2021-11-06:
It seems that the SKS keyserver has been deprecated, all you need to do is use an alternative key server. This worked for me:
`   sudo apt-key adv --keyserver keyserver.ubuntu.com  --recv-keys 00258F48226612AE`

Billir, 2021-11-06:
Thanks, I'll check it out.
Bill Rayment
Project Bee
1 403 630 4848

Billir, 2021-11-07:
Well it worked using Stretch Version and
Patrick's

It seems that the SKS keyserver has been deprecated, all you need to do is
use an alternative key server. This worked for me:
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys
00258F48226612AE

Got 3 sensors reading and graphing. It is shows only temperatures but no
graphs with Buster

Bill Rayment
Project Bee
1 403 630 4848

Billir, 2021-11-08:
Well I was wrong - it did present a graph with the temp range but did not
log. Sad it did work well in the past.
Bill Rayment
Project Bee
1 403 630 4848

DemBones78, 2021-11-09:
Yeah man just load home assistant on that raspberry pi you will be much
better off.   Hass.io

Alireza1369newi, 2021-12-12:
> Got this error when installing Temperature Machine 
> Executing: /tmp/apt-key-gpghome.Hd8dHFNGHw/gpg.1.sh --keyserver pool.sks-keyservers.net --recv-keys 00258F48226612AE
> gpg: keyserver receive failed: No name
> Any suggestions??

patrickn, 2021-12-12:
> > Got this error when installing Temperature Machine
> > Executing: /tmp/apt-key-gpghome.Hd8dHFNGHw/gpg.1.sh --keyserver pool.sks-keyservers.net --recv-keys 00258F48226612AE
> > gpg: keyserver receive failed: No name
> > Any suggestions??

Yes, SKS keyserver has been deprecated. See my answer above.

Billir, 2022-04-26:
It i now April 2022 and this key provided earlier still works 
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 00258F48226612AE

iw2ejh, 2022-12-15:
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 00258F48226612AE    is working now in decembre 2022, thanks