tobyweston/temperature-machine #115: Not possible to make the installation

simonmn2006, 2024-05-29:
Hi and thank you for sharing your project , im trying to install it but ther is a error with the apt-key

HenrysCat, 2024-05-29:
Try   sudo apt-key adv --keyserver keyserver.ubuntu.com http://keyserver.ubuntu.com/ --recv-keys 00258F48226612AE

simonmn2006, 2024-05-29:
Executing: /tmp/apt-key-gpghome.IElqaSTcQ1/gpg.1.sh --keyserver keyserver.ubuntu.com http://keyserver.ubuntu.com/ --recv-keys 00258F48226612AE
gpg: Note: '--recv-keys' is not considered an option
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
usage: gpg [options] [filename]

HenrysCat, 2024-05-29:
What OS are you using? a little more info will help

Also this project is completely dead now and has always been hard to set up and unrealiable.

simonmn2006, 2024-05-29:
hi sorry, pi os lite 32 bit